sandialabs/WecOptTool #4: scaling

ryancoe, 2021-06-14:
Since the forces and moments are often on the scale of 1e3 to 1e6 and velocities are on the scale 1e-1 to 1e0, the optimization problem needs some help with scaling... @cmichelenstrofer has already done this elsewhere, just need to apply here. Might be nice if you can make a note on this issue about the logic behind what you've done.

ryancoe, 2021-09-01:
[image]

ryancoe, 2021-09-01:
@gbacelli suggested a way to automatically set the scaling (or at least check if it is OK). We basically want to make sure the the motions and forces are of similar scales. We can check the gradient of the `f_i` and `f_{ext}` relative to `x`

[image]

ryancoe, 2021-11-18:
@cmichelenstrofer - Please record any resolution on the scaling question that we discussed this morning here for posterity.

cmichelenstrofer, 2021-11-25:
I ended up removing the `gi_block_scale` because it did not make sense to me to scale only one term in the dynamic equations and not the others.

ryancoe, 2021-11-29:
I'd like to reopen this and discuss a bit more. In the end, we want/need scaling.

cmichelenstrofer, 2021-11-29:
We still have scaling for the state variable (can scale WEC dynamics state and controller state separately) and for the objective function.

ryancoe, 2021-11-30:
Ahhh, ok. Thanks.

closed 2021-11-30 by ryancoe

ryancoe, 2022-03-10:
In #48, we introduced a "heads up display" the outputs the mean values of `x_wec`, `x_opt`, and the objective function. This isn't necessarily the perfect thing to base the scaling on. Take, e.g., a regular wave, where all of the components except one or two are zero.

PR #10 (2021-11-28): restructuring code.
- [x] Figure out bug 
- [x] Reproduce the Case A results. 
- [x] Figure out way to do multiple DOFs that works with Autograd
- [x] Test cases: 1- WaveBot multiple DOFs, 2- WaveBot multiple PTOs, 3- Two wave directions, 4- irregular waves, 5- RM3 multiple bodies and generalized DOFs
- [x] Cleanup (once above tasks completed, code in somewhat final shape): docstrings, logging, naming convention, etc.
- [x] Modify the tests to to work with new code format

Closes #2. Closes #4. Closes #5. Closes #6. Closes #7. Closes #12.