sandialabs/WecOptTool #6: Wave directionality

ryancoe, 2021-06-14:
Our current tensor structures are set up to handle multiple modes of motion, which may arise from a multi-modal device and/or multiple devices within an array. However, we do not currently have a means of handling directional waves. This should be simple enough however and relatively undisruptive to introduce, since the additional dimension needed to handle wave directionality will only show up when calculating the excitation force, after which this dimension is projected into the modes of motion.

ryancoe, 2021-06-18:
@cmichelenstrofer and I were discussing this more. Key step per this issue is going from `\hat{A}(\omega, \theta)` and `\hat{H}_{ex}` to `\hat{F}_{ex}`.

```latex
&n\textrm{: number of DOFs}\\
&m\textrm{: number of frequencies}\\
&p\textrm{: number of directions}\\
&\textrm{Directional wave spectral density: }S(\omega, \theta) \in \mathcal{R}^{m \times p}\\
&\textrm{Complex amplitude spectrum: }\hat{A}(\omega, \theta) \in \mathcal{C}^{m \times p}\\
&\textrm{Excitation transfer function: }\hat{H}_{ex} \in \mathcal{C}^{n \times m \times p}\\
&\textrm{Complex excitation spectrum: }\hat{F}_{ex} \in \mathcal{C}^{n \times m}\\
&\textrm{Intrinsic impedance: }\hat{Z}_i \in \mathcal{C}^{n \times n \times m}\\
&\textrm{Complex state: }\hat{x} \in \mathcal{C}^{n \times m}
```

[image]

cmichelenstrofer, 2021-11-25:
PR #10 includes wave directionality.

PR #10 (2021-11-28): restructuring code.
- [x] Figure out bug 
- [x] Reproduce the Case A results. 
- [x] Figure out way to do multiple DOFs that works with Autograd
- [x] Test cases: 1- WaveBot multiple DOFs, 2- WaveBot multiple PTOs, 3- Two wave directions, 4- irregular waves, 5- RM3 multiple bodies and generalized DOFs
- [x] Cleanup (once above tasks completed, code in somewhat final shape): docstrings, logging, naming convention, etc.
- [x] Modify the tests to to work with new code format

Closes #2. Closes #4. Closes #5. Closes #6. Closes #7. Closes #12.
closed 2021-11-28 by ryancoe