sandialabs/WecOptTool #7: Move to capytaine 1.2.1

ryancoe, 2021-06-14:
I help make some minor updates to `capytaine` that are part of [v1.2.1](https://github.com/mancellin/capytaine/releases/tag/v1.2.1). In particular, there was a bug about the naming of the mass matrix. We should use this latest version of `capytaine` in `WecOptTool`.

changelog: https://github.com/mancellin/capytaine/blob/master/docs/changelog.rst

@cmichelenstrofer - as a general mode of operating, it may often sense to make alterations directly to `capytaine` instead of in `WecOptTool` - I definitely support this

cmichelenstrofer, 2021-06-14:
The `hydrostatic` module in `capitaine` changed significantly, need to update WEC-Opt tool to use this new module.

ryancoe, 2021-07-08:
Another note is that `capytaine` 1.2.1 has a method for calculating the impedance: https://github.com/mancellin/capytaine/blob/master/capytaine/post_pro/impedance.py

cmichelenstrofer, 2021-11-11:
PR  #10 uses `capytaine` v1.3.0

PR #10 (2021-11-28): restructuring code.
- [x] Figure out bug 
- [x] Reproduce the Case A results. 
- [x] Figure out way to do multiple DOFs that works with Autograd
- [x] Test cases: 1- WaveBot multiple DOFs, 2- WaveBot multiple PTOs, 3- Two wave directions, 4- irregular waves, 5- RM3 multiple bodies and generalized DOFs
- [x] Cleanup (once above tasks completed, code in somewhat final shape): docstrings, logging, naming convention, etc.
- [x] Modify the tests to to work with new code format

Closes #2. Closes #4. Closes #5. Closes #6. Closes #7. Closes #12.
closed 2021-11-28 by ryancoe