sandialabs/WecOptTool #11: Parallel Computation

ssolson, 2021-08-03:
https://github.com/mancellin/capytaine/issues/65#issuecomment-884965605

@ryancoe I was able to see the use of all of my cores when running the capytaine example. 

Using the current WecOptTool main repository ` example_WaveBot_oneDof.py` I am not sure the problem is difficult enough to view the issue. Can you tell what specifically you were running/ testing in WecOptTool to test this?

ryancoe, 2021-08-04:
As long as you are asking for multiple _problems_ (e.g., multiple frequencies, multiple DOFs -- all of which are solved independently in BEM), you should see it using multiple cores. Thus, because `example_WaveBot_oneDof.py` looks at 18 frequencies, you should see it using multiple cores.

https://github.com/SNL-WaterPower/WecOptTool/blob/00770b41537f2bea0bb01ee9dc0e578d98382e89/WecOptTool/examples/example_WaveBot_oneDof.py#L28

ryancoe, 2021-12-14:
As far as I can tell, parallel computation for `capytaine` is working properly for me now (macOS). Going to close for the time being.

closed 2021-12-14 by ryancoe