sandialabs/WecOptTool #12: psuedo-spectral optimization algorithm/package

ryancoe, 2021-09-10:
We are currently using [`scipy.optimize.minimize`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.minimize.html).

We have previously played with using IPOPT for the pseudo-spectral solution. From what I can tell, this is developed using C++ by [COIN-OR](https://www.coin-or.org); there are a couple of Python wrappers.

ryancoe, 2021-09-10:
It looks like this conda package might be an option: https://anaconda.org/conda-forge/ipopt

cmichelenstrofer, 2021-11-11:
I was able solve the optimization  problem with either `scipy.optimize.minimize` or `ipopt` via the [cyipopt](https://github.com/mechmotum/cyipopt) wrapper. The optimization package was not the issue (see #10). Since `scipy` does just fine, we will stick with it for now. If we want we can add a flag to switch between optimizers in the future.

ryancoe, 2022-06-07:
Re-opening this as we really should take another look at IPOPT. I believe the best option for using this in Python is [`cyipopt`](https://github.com/mechmotum/cyipopt)

cmichelenstrofer, 2022-06-07:
*__Pyomo__* is a Sandia developed code (moved from internal to public GitHub in 2016) that is becoming very popular (actively maintained/developed, weekly developer cals open to the community, thousands of users, some big projects using it). I have been meaning to check it out. Not sure if it is a good option for this project.

- [Github](https://github.com/Pyomo/pyomo)
- [Website](https://www.pyomo.org/)

cmichelenstrofer, 2022-10-20:
For now `scipy` works well, and there is no immediate plan to address this. I am moving it to a discussion (#140).

rebeccamccabe, 2023-02-14:
I am wondering if you have considered [pyoptsparse](https://github.com/mdolab/pyoptsparse) - it contains wrappers for IPOPT, SNOPT, and various other solvers with a single interface, and is the primary optimization driver for OpenMDAO, which I plan to use in my project. (Separately, I'm still figuring out a strategy for using WOT with OpenMDAO because the nested optimization structure makes it a "collaborative optimization" which requires a different setup.) But perhaps pyoptsparse would be helpful for WOT in the future, while also potentially facilitating OpenMDAO integration.

cmichelenstrofer, 2023-02-14:
@rebeccamccabe I wasn't aware of it but seems like a good option!

ryancoe, 2023-11-10:
@rebeccamccabe - Thanks very much for this suggestion! Briefly looking into this a bit more... [`pyoptsparse`](https://github.com/mdolab/pyoptsparse) does look interesting, but digging a bit deeper it looks like some of the methods we're immediately interested in using (e.g., `IPOPT`) require separate installation, often via compiling on your machine. This is definitely something that would discourage us from using the package, as our average user would likely be challenged to get the install done right. Maybe it would work fine if we identify a method within [`pyoptsparse`](https://github.com/mdolab/pyoptsparse) that works well for our problem and does not require a complex installation. I'd definitely like to see if there's a better option out there for us, so please keep the suggestions coming.

ryancoe, 2023-11-21:
I was doing some reading tonight on a related topic and looked again at options for IPOPT Python interfaces. It looks like [`cyipopt`](https://github.com/mechmotum/cyipopt) has been under very steady development for a while now and has a _scipy-style_ interface. I was able to install it and run the examples. I was even able to install [the CPU jaxlib](https://github.com/google/jax#installation) and run an example where that's used to provide the jacobian! I think this might be a winner. @cmichelenstrofer @michaelcdevin - Any thoughts?

cmichelenstrofer, 2023-11-21:
sounds good to me! Would we gain any new capability?

gbacelli, 2023-11-21:
yes, it comes with chatGPT... 😀

ryancoe, 2023-11-21:
> Would we gain any new capability?

@cmichelenstrofer - Not entirely sure... when @gbacelli and I were planning the restructuring of WecOptTool (when we transitioned from MATLAB to Python) we got some advice from an optimization expert. I remember two things in particular:

1. ~~Use some sort of autodiff~~ (done)
2. Consider IPOPT

It'd probably be best to find some problem(s) where we struggle (e.g., the convergence is sensitive to the scaling factors) and swap `cyipopt` for `scipy` to see if there's any improvement. Now that the installation for `cyipopt` is straightforward and it uses the same structured interface as `scipy`, I think this should be a relatively quick task.

gbacelli, 2023-11-21:
FYI: Interior point methods (the ones implemented in IPOPT) are the preferred solutions for numerical optimal control problems.

michaelcdevin, 2023-11-22:
`cyiopt` looks great, good find @ryancoe!

My only hesitation is that `cyiopt` has an EPL-2.0 license, which has a derivative works clause. WecOptTool is currently licensed under GPL-3.0. However, I believe EPL is less restrictive than GPL with what constitutes a "derivative work"; I think if we use `cyiopt` without modification, we can continue using a GPL-3.0 license without issue. We should verify this before making any changes though.

rebeccamccabe, 2023-11-27:
> @rebeccamccabe - Thanks very much for this suggestion! Briefly looking into this a bit more... [`pyoptsparse`](https://github.com/mdolab/pyoptsparse) does look interesting, but digging a bit deeper [...]

There is a provided script to handle the build and install of ipopt for you in pyoptsparse https://github.com/OpenMDAO/build_pyoptsparse/. Some of the pyoptsparse features might be handy like parallelism, easily swapping solvers, history, and open MDAO driver, though I don’t see a problem with cyipopt either. Pyoptsparse having an open MDAO driver doesn’t matter nearly as much as I thought it would because “At this time, OpenMDAO does not evaluate gradients through the optimization process, so a gradient-based driver at the outer level cannot access derivatives across the optimization on the inner level.“ So it would be helpful if you wanted to run WecOptTool via open mdao in the future, ie make subsystems for hydro and geometry and PTO, but not helpful for a nested optimization (which is likely the structure of my project).
(https://openmdao.org/newdocs/versions/latest/features/building_blocks/components/submodel_comp.html). 
Pyoptsparse doesn’t have auto diff, though open mdao has some jax integration.

cmichelenstrofer, 2023-11-27:
When making the changes to `cyiopt` + `jax` we should think about if we can create a mechanism for easily changing the optimization package. I think it would only require allowing to overload a small part of the `WEC.solve` function.

rgcoe, 2025-05-29:
I successfully installed `pyoptsparse` and then `IPOPT` per the [`build_pyoptsparse`](https://github.com/OpenMDAO/build_pyoptsparse/) instructions as suggested by @rebeccamccabe  in [her comment above](https://github.com/sandialabs/WecOptTool/issues/12#issuecomment-1827412613) on my M1 mac and then updated [`rosenbrock.py` example](https://github.com/mdolab/pyoptsparse/blob/main/examples/rosenbrock.py) to use `IPOPT` instead of `SQLP` and got the same result. Nothing amazing, but if our main concern about using `pyoptsparse` to utilize `IPOPT` is user-friendly installation, it might be useful for @dtgaebe and/or @jtgrasb to try this and see if they have similar success on Windows.

```bash
mamba create -n pyoptsparse pyoptsparse
mamba activate pyoptsparse
git clone git@github.com:OpenMDAO/build_pyoptsparse.git
cd build_pyoptsparse
python -m pip install .
cd ..
git clone git@github.com:mdolab/pyoptsparse.git
cd pyoptsparse/examples
python rosenbrock.py
# edit rosenbrock.py that 'opt = IPOPT(options=optOptions)'
python rosenbrock.py
```

dtgaebe, 2025-08-11:
> I successfully installed `pyoptsparse` and then `IPOPT` per the [`build_pyoptsparse`](https://github.com/OpenMDAO/build_pyoptsparse/) instructions as suggested by [@rebeccamccabe](https://github.com/rebeccamccabe) in [her comment above](https://github.com/sandialabs/WecOptTool/issues/12#issuecomment-1827412613) on my M1 mac and then updated [`rosenbrock.py` example](https://github.com/mdolab/pyoptsparse/blob/main/examples/rosenbrock.py) to use `IPOPT` instead of `SQLP` and got the same result. Nothing amazing, but if our main concern about using `pyoptsparse` to utilize `IPOPT` is user-friendly installation, it might be useful for [@dtgaebe](https://github.com/dtgaebe) and/or [@jtgrasb](https://github.com/jtgrasb) to try this and see if they have similar success on Windows.
> 
> mamba create -n pyoptsparse pyoptsparse
> mamba activate pyoptsparse
> git clone git@github.com:OpenMDAO/build_pyoptsparse.git
> cd build_pyoptsparse
> python -m pip install .
> cd ..
> git clone git@github.com:mdolab/pyoptsparse.git
> cd pyoptsparse/examples
> python rosenbrock.py
> \# edit rosenbrock.py that 'opt = IPOPT(options=optOptions)'
> python rosenbrock.py

@rgcoe I tested your installation instructions on windows (using ```conda``` instead of mamba and https to clone the repos). 

I tested:
``` 
python ronsebrock.py
3.231878358118566e-07
# change rosenbrock.py
# # First party modules
# from pyoptsparse import SLSQP, Optimization, IPOPT
# # Create optimizer
# # opt = SLSQP(options=optOptions)
# opt = IPOPT(options=optOptions)
python rosenbrock.py
7.320416311460178e-08
```

The solve with ```IPOPT``` took significantly longer, but other than that pretty smooth and seems to be working on windows.

PR #10 (2021-11-28): restructuring code.
- [x] Figure out bug 
- [x] Reproduce the Case A results. 
- [x] Figure out way to do multiple DOFs that works with Autograd
- [x] Test cases: 1- WaveBot multiple DOFs, 2- WaveBot multiple PTOs, 3- Two wave directions, 4- irregular waves, 5- RM3 multiple bodies and generalized DOFs
- [x] Cleanup (once above tasks completed, code in somewhat final shape): docstrings, logging, naming convention, etc.
- [x] Modify the tests to to work with new code format

Closes #2. Closes #4. Closes #5. Closes #6. Closes #7. Closes #12.