sandialabs/WecOptTool #18: BUG: Funky read_bem errors

ryancoe, 2021-12-16:
**Describe the bug**
Changes to mesh (and other things) can cause the usage of `read_bem` to throw an unhelpful error message.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `tutorial_3_optimization.ipynb`
2. Adjust the `mesh_size_factor`
3. Re-run the notebook
4. This will throw the assertion error from the following line

https://github.com/SNL-WaterPower/WecOptTool/blob/19a523f65165a6ef7544613ab1c8336db0dd96d1/wecopttool/core.py#L391

**Expected behavior**
Better error message (e.g., "Did you change you mesh?")

**System:**
 - OS: macOS
 - Python version: 3.9.7
 - WecOptTool version: 1.0.0
 - Any other relevant package version: N/A

**Additional information**
Suggested solution: wrap the assertions in `read_bem` within a `try-catch` environment and add a more helpful error message when any of them fail

cmichelenstrofer, 2021-12-17:
The problem in your case is that the new calculated mass values do not match those used for the saved BEM solution (due to mesh change). I agree we could add a more helpful error message. Note that this assertion can fail if the mass changed for any reason, not just because of different meshes. The `assert` statements accept an error message as `assert <condition>, <message>`. This might be sufficient (rather than `try-catch`).

ryancoe, 2021-12-17:
Total agree that fully using the `assert` statement is better than `try-catch`

ryancoe, 2022-03-01:
I looked at this again just now and could not reproduce the error message. The changes with #28 seem to have fixed this. Closing for now.

closed 2022-03-01 by ryancoe