sandialabs/WecOptTool #35: Feature request: cache deps. for CI/CD

ryancoe, 2022-01-24:
https://docs.github.com/en/actions/advanced-guides/caching-dependencies-to-speed-up-workflows

cmichelenstrofer, 2022-10-27:
Now caching the conda environment with capytaine and wavespectra (and all their dependencies, e.g. numpy, scipy, etc). This will be part of v2.1. When a cache is available this reduces the time to install in ubuntu from ~1.5 minutes to ~15 seconds. On MacOS and Windows there is significant speedup but install still can take long (~5 minutes on Windows!). On Ubuntu almost the netirety of the run time are the tests (~2 minutes).

cmichelenstrofer, 2022-10-27:
The cache is updated:

- if it is the first time the tests are run that calendar day
- if the `pyproject.toml` file has changed
- if you manually change the `CACHE_NUMBER` value

commit eff21899 (2022-10-31): OTHER: CI/CD improvements (#151)

* `wavespectra` now installed by default
* remove `CHANGES.md` file, use GitHub release notes instead.
* cache dependencies Fixes #35
* simplify number of files and content and add comments explaining each step
* GH pages only deployed on release
* Update `README.md`, `CONTRIBUTING.md`, etc.
* Other minor CI/CD changes
closed 2022-10-31 by cmichelenstrofer