sandialabs/WecOptTool #52: Validate multiple PTO DoF's

cmichelenstrofer, 2022-02-21:


dtgaebe, 2022-02-23:
Thank you for bringing this up! I have to take back what I said earlier. I think we are _mostly_ doing it right. My current understanding is that:

The way I derived the forward kinematics is fine to relate PTO forces to the generalized force vector on the WEC with the Jacobian  _J_ ,
 <img src="https://render.githubusercontent.com/render/math?math=F_{p}= J f_p">
The **"inverse" kinematics**, i.e. from WEC _x_ to PTO _p_ frame_ are in **fact the transpose of the Jacobian _J_**, 
<img src="https://render.githubusercontent.com/render/math?math=\dot{p}= J^{T} \dot{x}">,
BUT the inverse kinematics, are not actually the inverse in terms of linear algebra. Commonly the literature states uses J^-1 to denote the inverse kinematics, but the way we are using the relation 
<img src="https://render.githubusercontent.com/render/math?math=J^{T} J \ne I">

One **issue** in the code is the **position method**. If I understand it correctly, we can use the Jacobian to relate forces and velocities in the different frames, but not the position coordinates.

I think we need to
1. use the WEC position to calculate the Jacobian <img src="https://render.githubusercontent.com/render/math?math=J(x)">
2. The PTO positions are either constant (PTO outside of WEC), or directly dependent on the WEC position (if PTO on WEC)
3. use the transpose Jacobian to calculate PTO velocities <img src="https://render.githubusercontent.com/render/math?math=\dot{p}= J^{T}(x) \dot{x}">
4. use the Jacobian to convert PTO force to generalized force vector on the WEC  <img src="https://render.githubusercontent.com/render/math?math=F_{p}= J(x) f_p">

cmichelenstrofer, 2022-02-23:
@dtgaebe what we want is the PTO position not velocity, so if I understand your explanation correctly we in fact need something different than just `J^T`. But maybe we could create this transform from `J`  followed by an integration. 

The conclusion that the transformation from WEC velocity to PTO velocity is simply the transpose of the Jacobian is general? For any number and type of WEC and PTO DOFs? Could you document this?

cmichelenstrofer, 2022-02-23:
@ryancoe it sounds like there is a serious problem in the code, but that the solution would not require more information from the users as I thought.

cmichelenstrofer, 2022-02-23:
@dtgaebe if `J` and `J^T` are not inverse (`J^T J =/= I`) then what does the transformation from `p_dot` to `x_dot` look like? It can't be `x_dot = J p_dot`. Also how do you type math in here?

dtgaebe, 2022-02-23:
@cmichelenstrofer I see. In that case, the PTO positions should always be a linear combination of the WEC state, i.e. can be obtained by rotation and translation with the standard homogeneous transformation matrix
[image]
This holds for the general case of 6Dof, but requires rotation about the other Dofs as well.

Yes, the transpose should be generally applicable, but I will look more into this. Yesterday I found a paper that uses the inverse kinematics for a tether submerged WEC and it turned out to be the transpose of what I derived for the moored WECs. https://www.sciencedirect.com/science/article/pii/S0960148116301495 
But I can look into the robotics, plain kinematics literature again and document it!

I see you point and don't know the answer. I will need to look into the theory again....
Here is a full thread with hacks on how to write math: https://gist.github.com/a-rodin/fef3f543412d6e1ec5b6cf55bf197d7b
But I just use the simplest, albeit not very elegant, way: 
`<img src="https://render.githubusercontent.com/render/math?math=e^{i \pi} = -1"  >`

cmichelenstrofer, 2022-02-23:
To clarify

- the PTO does not have to be a line going from the WEC to an anchor at the seabed. E.g. the RM3 could have a direct drive, with the PTO motion equal to the buoy heave - spar heave.  We want a general answer for any PTO and WEC archetype.
- The question boils down to **"Are the two transformations we need related to each other? I.e. can we get one given the other?"** for arbitrary WEC and PTO archetypes. The two transformations are:
    - From PTO force (in the PTO frame) to forces on the WEC 
    - From WEC position to PTO position (in PTO frame)

dtgaebe, 2022-02-24:
Yes, but we always need the **direction** that the PTO is acting in. If the direction is the difference (in Cartesian coordinates in the inertial frame) between an anchor and PTO, or the difference between a point on the WEC body and the PTO doesn't matter for the transformation of the forces. This transformation is the Jacobian, which generally depends on the positions of the PTO force attachment point, which depends on the WEC instantaneous position. (The transformation (Jacobian) will be different for rotational PTOs )

- I think that the transformation from WEC position to PTO position is easy. It is eq. (8) in my previous post.
    - Why do we need the PTO position in the PTO frame again? Shouldn't that be constant?
- The Jacobian to relate the forces from PTO to WEC depends on
    - The direction that the PTO is acting in
         - Which depends on the PTO position in the inertial frame and a reference point also in the inertial frame (thus eq. (8))
     - The position of the PTO attachment relative to the bodies center of gravity (I'd use the constant PTO position in the bodies coordinate system)
         - Which is required for the torques on the WEC
     - For a rotational PTO we need the axis of rotation

I did quite a bit of theory on paper today and will type it up soon, should be more elegant now. Long story short, I think the user will only have to provide the reference points and decide between linear, or rotational PTO. 
The coordinate transformation for the positions only depends on the instantaneous WEC position, we'll use that to compute the Jacobian, which we will use for the relation between PTO force in PTO frame and the force acting on the WEC.

cmichelenstrofer, 2022-02-24:
* the PTO motion is not constant, else there is PTO no force or power. 
* the attachment points can be between different WEC bodies.

cmichelenstrofer, 2022-02-25:
I took a stab at this after @dtgaebe pointed me towards using conservation of energy in both frames. This was the additional information that was needed. I am know convinced we are doing it correctly. See figure below. @dtgaebe can you double check that this looks correct and I will close the issue. 

[image]

dtgaebe, 2022-02-25:
Yes, that is it! But _p_  and _x_ are velocities.

Per this definition the kinematic matrix we need to ask the user for is commonly referred to as the **inverse kinematics**! That's where I was wrong in trying to focus on the standard Jacobian (forward kinematics from PTO to endeffector (WEC))

Last question I have is about the position method in the PTO class. It is also computed with the K^T, which I don't understand.

dtgaebe, 2022-02-25:
No, my apologies for the confusion. 

Your math is correct, but now I am again confused how _K_ looks like, or can be derived.

cmichelenstrofer, 2022-03-01:
There is no error in the code! I did make some changes to make the kinematics more clear (#63)

closed 2022-03-01 by cmichelenstrofer