sandialabs/WecOptTool #57: Validate irregular waves (directional and multidirectional)

cmichelenstrofer, 2022-02-22:
Create validation cases that use irregular waves, both unidirectional and multidirectional. Compare to linear solution.

dtgaebe, 2022-03-11:
@cmichelenstrofer  Currently, regular_wave() allows the user to specify multiple directions. Should we implement a prevention?

cmichelenstrofer, 2022-03-15:
@dtgaebe something like this should do the trick:
```python
# check that only a single direction is given
assert len(np.atleast_1d(np.array(direction).squeeze()))==1
```

ryancoe, 2022-03-22:
@dtgaebe - Seems like this could logically get included in #83. Do you think you could do something like:

```python
t_dom, f_dom , _ = wec.solve(multi_dir_wave)
sol_analytic = np.abs(Fe)**2 / (8*np.real(Zi))
assert f_dom == sol_analytic
```

dtgaebe, 2022-03-22:
Yes, I'm currently working on this, or rather I was trying to use the `power_limit()` from `core.py`

ryancoe, 2022-03-22:
Great! Per #69, we are also not sure if `power_limit()` works correctly for multiple DOFs.

dtgaebe, 2022-04-22:
cosine2s spread function reference:

https://apps.dtic.mil/sti/pdfs/ADA591687.pdf

closed 2022-10-17 by cmichelenstrofer