sandialabs/WecOptTool #60: Feature request: Construct PTO Jacobian for rigid body motions

cmichelenstrofer, 2022-02-25:
**Feature description.**
If the WEC DOFs are translational and rotational rigid body motions we can help the user construct the Jacobian matrix by asking them the type of PTO (translation, rotation) & the rotation axis for rotational. @dtgaebe has derived these equations. This would fit nicely in the PTO module.

**Issue addressed.**
Make it easier for users to use our PTO classes. Currently they have to calculate the Jacobian matrix and provide that as input, but we can easily construct this matrix for them.

**Describe the solution you'd like**
A function in the PTO module that users can optionally use.

**Describe alternatives you've considered**
Currently creating the Jacobian falls completely on the user.

**Interest in leading this feature development?**
@dtgaebe is interested in leading this.

**Additional information**

cmichelenstrofer, 2022-04-13:
This paper should help: https://www.sciencedirect.com/science/article/pii/S0960148116301495

dtgaebe, 2022-04-13:
Yes, that's what I have implemented in matlab.
And poorly here. I will get to this soon

cmichelenstrofer, 2022-04-13:
@dtgaebe Cool! I wasn't aware of this paper and it came up in a meeting today, so thought I would share.