sandialabs/WecOptTool #62: Feature request: Add linear force components to post-processing.

cmichelenstrofer, 2022-02-28:
**Feature description.**
Calculate the different force components for the optimized solution (using the optimal state + the BEM coefficients) and include it in the xarray in the post-processing function. These forces include: excitation (split into diffraction + F-K?), radiation damping, hydrostatic, added mass.

ryancoe, 2022-03-02:
Along with this, we might also want to allow users to do something like the following. Could be handy, but don't want to add too much bloat.

```python

additional_forces_to_post_process_only = {'Tether tension': -1 * tether_force_on_WEC, 
                                          'support arm bending moment': -1 * tether_force_on_WEC * moment_arm}

my_wec = wot.WEC(..., additional_forces_to_post_process_only=additional_forces_to_post_process_only)

sol_time_dom, = my_wec.solve(...)

sol_time_dom[ 'support arm bending moment'].plot()
```

cmichelenstrofer, 2022-10-20:
completed/fixed in v2.0

closed 2022-10-20 by cmichelenstrofer