sandialabs/WecOptTool #72: Natural frequency in multiple DOFs

ryancoe, 2022-03-07:
Similarly to #69, we need to verify the functionality of our `natural_frequency` method and function in multiple degrees of freedom.

https://github.com/SNL-WaterPower/WecOptTool/blob/7d9bbc70f7cac23b468d5b073d6da9727c68be9e/wecopttool/core.py#L1105-L1127

cmichelenstrofer, 2022-10-20:
This didn't make it into v2.0. v2.1 is planned mostly for CI/Testing. Will leave this open as a good first issue.

ryancoe, 2023-05-15:
Note that both [`tutorial_3_LUPA.ipynb`](https://github.com/sandialabs/WecOptTool/blob/main/examples/tutorial_3_LUPA.ipynb) and #214 present good opportunities to test this functionality.

jtgrasb, 2024-12-24:
This would be a pretty easy thing to add back in to utilities.py if we want to.