sandialabs/WecOptTool #74: Better scaling and/or initial conditions

ryancoe, 2022-03-10:
Do one or both of the following:

 - Run w/o inequality constraints
 - find the analytical solution (v = Fe/2/Ri)

From these, you can do one or both of the following:

- set `x0`
- set scaling

related to #4 #48

ryancoe, 2022-03-24:
Note that if we use the **_find the analytical solution (v = Fe/2/Ri)_** approach as suggested above, we would ideally do this while considering the PTO impedance as being developed in #38. However, we cannot necessarily be sure how the user's PTO will look in all cases. Thus we probably want to stick with finding the analytical solution for optimal mechanical power transfer and using this for the initial guess and scaling (in the absence of input from the user). It is definitely not perfect, but probably better than nothing at all (we can/should check whether this is indeed true).

ryancoe, 2022-03-25:
From [Falnes](https://books.google.com/books/about/Ocean_Waves_and_Oscillating_Systems.html?id=bl1FyQjCklgC&source=kp_book_description), eq. 3.46:

[image]

ryancoe, 2022-03-25:
@cmichelenstrofer - I see we have a function for going from a sin and cos components (i.e., our pseudo-spectral components) to a complex spectrum ([`real_to_complex_amplitudes`](https://github.com/SNL-WaterPower/WecOptTool/blob/6f11bd30fa0fcde98a0c67ecb41b8407a20fe381/wecopttool/core.py#L894-L904)), do we have anything for doing the opposite?

cmichelenstrofer, 2022-03-25:
We do not, but it would be a good addition and easy to implement.

ryancoe, 2022-03-29:
I've done a couple versions of this in #89. I'm going to note the successes/failures here for posterity:

- **using analytic hydrodynamic solution for guess and scaling:** I was only able to use this to set an initial guess for `x_wec_0`. Because it does not give any information on `x_opt` or `obj`, you cannot use this to set scaling and the effectiveness of setting an initial guess for only part of the solution may be limited. Nonetheless, I have added functions for the user to do this if helpful.

- **run w/o inequality constraints first, then add constraints back:** This appears to have mixed effectiveness. In the WaveBot case that I added to our tests, this approach does appear to improve the solution speed. However, when I tried to apply this to our examples, I did not see the same improvements.

**In summary, neither of these approaches is a silver bullet, but hopefully do add some benefit to users.**

cmichelenstrofer, 2023-09-28:
Becca and Maha suggested using the Hessian to set the scale. They will share some code. The Hessian and scales would depend on the initial guess.

rebeccamccabe, 2023-10-07:
My implementation of hessian scaling (in matlab) is https://github.com/symbiotic-engineering/MDOcean/blob/code-cleanup/mdocean/optimization/run_solver.m. The method came from Pappalambros & Wilde Principles of Optimal Design, section 8.3. I run the unscaled optimization for a few (8) iterations, and get the hessian from that point, and then continue optimizing. My code gets the hessian from finite difference, but for WecOptTool you can perhaps get it from the autodiff.

ryancoe, 2024-04-03:
We looked into this a bit more today... a couple of notes:

### Hessian
For an unstructured controller, the second derivative (diagonal of the Hessian) of our objective function (e.g., mech. power) is always zero.

$$
\frac{\delta ^2 P}{ \delta F^2} = \frac{\delta ^2 P}{ \delta v^2} \equiv 0
$$

Note, however, that mixed second order partial derivatives can be non-zero.

$$
\frac{\delta ^2 P}{ \delta v \delta F} \not\equiv 0
$$

Note also that if you were using a _structured_ controller (e.g., PI), you would have some non-zero elements on the diagonal of Hessian.

### Initial guess
While it seems possible that for specific problems you can come up with better initial guess strategies, a crude workable solution appears to be using the wave amplitude spectrum to give an initial guess. Logically, this would give you a good answer for a "wave follower" body. Initial experiments with this idea look promising.

cmichelenstrofer, 2024-04-03:
From #225: the average mechanical power is 

$$\bar{P}_{mech} = \pi \sum _{n=1}^{N-1}\biggl(n \left(f _{a,n} x _{b,n} - f _{b,n} x _{a,n} \right)\biggr) $$

so 

$$ \frac{\partial \bar{P}_{mech}}{\partial^2 x _{a,n}} = 0 $$ 

similarly for $x _{b,n}, f _{a,n}, f _{b,n}$

and 

$$\frac{\partial \bar{P}_{mech}}{\partial x _{b,n}\partial f _{a,n} } = \pi n$$

and similar for the other cross terms. 

Numerically I get this linear behavior but with a slope smaller than $\pi$... need to investigate further. 

 Hessian (50 frequencies, x = [x_wec, x_opt]):

[image]

Lower diagonal: 
[image]

cmichelenstrofer, 2024-04-03:
@rebeccamccabe we found out that the diagonal of the Hessian is zero when using an unstructured controller (see examples above). What type of problem did you use the Hessian for? Was it a structured controller?

rebeccamccabe, 2024-04-04:
My problem was not WecOptTool or pseudospectral at all, the design
variables were geometric dimensions not control/dynamics harmonics, and the
objective was LCOE not power, so the hessian was nonzero in most elements
of the matrix.

dtgaebe, 2024-04-04:
Analog to @cmichelenstrofer's comment above, but using a PI controller the force components become a function of the dynamic states. Consequently, the Hessian does have non-zero values on it's diagonal. Let's focus on the components inside the freq array (*not* DC and Nyquist, $n \in [1, nfreq-1]$)

$$f_{a,n} = K_p x_{b,n} \omega_n + K_i x_{a,n}, \quad f_{b,n} = -K_p x_{a,n} \omega_n + K_{i} x_{b,n}.$$

And the mechanical power components 

$$P_{a,n} = f_{a,n} \omega_n x_{b,n} , \quad P_{b,n} = f_{b,n}(- \omega x_{a,n}) $$

have non-zero values for the Hessian

$$\frac{\partial P_{a,n}}{\partial x_{b,n}} =  \frac{\partial P_{b,n}}{\partial x_{a,n}} = K_p \omega_n^2$$

i.e. the Hessian is a quadratic function of the radial frequency scaled by the proportional gain
[image]

The numeric values for the Hessian come in pairs, when taking every other component, we can numerically confirm that the Hessian looks as we would expect.
[image]

rebeccamccabe, 2025-01-15:
I was writing up my Hessian use for a paper and remembered this thread. I wanted to clarify that I am using the Hessian of the Lagrangian, which is the Hessian of the objective plus the lagrange multipliers times the Hessian of the constraints, see ie equations here https://www.mathworks.com/help/optim/ug/hessian.html#bsapedt. I'm not sure whether it is theoretically better for scaling to use H of objective or H of Lagrangian. I am just using Lagrangian since that is the only one matlab outputs automatically.

The Hessian of the Lagrangian in wecopttool could be nonzero even when the Hessian of the objective is zero (unstructured controller), if there are any user defined added forces or nonlinear dynamics. For linear dynamics, I think the dynamics equality constraint Hessian would be zero, since a linear expression has just the first derivative nonzero. But nonlinear dynamics + PTO/other constraint might have a nonzero  Hessian that might be helpful to include.

ryancoe, 2025-01-20:
Thanks for the update @rebeccamccabe. Please share your paper with us when you're ready!

PR #89 (2022-04-08): Initial guess, automatic scaling, multi-DOF power limit 
## Description
 - [x] Fixes #74
 - [x] Fixes #69

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/SNL-WaterPower/WecOptTool/blob/main/.github/CONTRIBUTING.md)
- [x] The pull request is from an issue branch (not main) on your fork to the [main branch in WecOptTool](https://github.com/SNL-WaterPower/WecOptTool). 
- [x] The authors have given the admins edit access
- [x] All changes adhere to the style guide including PEP8, Docstrings, and Type Hints.
- [x] Added note in [Changelog](https://github.com [... cut]