sandialabs/WecOptTool #75: Don't allow user to bound x_wec

ryancoe, 2022-03-10:
With #71, the user can set bounds on the decision variable. We may want to only allow the user to set bounds on `x_opt`, not `x_wec`. Not entirely sure about this... let's hash it out here.

ryancoe, 2022-03-10:
@cmichelenstrofer and I decided to expose the bounds as `bounds_wec` and `bounds_opt`.

ryancoe, 2022-03-23:
@cmichelenstrofer - I thought more about this as I was implementing in #78... I think I prefer to leave the `bounds` argument as is (i.e., don't split it out into separate `bounds_wec` and `bounds_opt` components. The fewer lines of wrapper code that we write for `scipy.optimize.minimize` the better. The only benefit to splitting these into separate args is to prevent user mistakes. I think given the state of our code (fairly immature), it is going to require an advanced user to be exercising this feature and they probably don't need our hand holding. If you agree, I will close this for now without implementing the change; we can revisit later if we like.

cmichelenstrofer, 2022-03-23:
I think this is important and we should leave the issue open, but I agree that it is not a priority. Reasons why they should be separate for `x_wec` and `x_opt`:

- Everything else is separated: `x0`, `scale_x`, function inputs. We never ask the user to deal with the combined state (that is all internal). 
- The use of these bounds for `x_wec` and `x_opt` are fundamentally different. It will be common to only want to bound one or the other. 

I would say you can commit #78 without this and we'll deal with this later.

ryancoe, 2022-03-24:
I feel such conviction for my standpoint, but then somehow you convince me! I'll complete this in #78.

PR #78 (2022-04-05): treat bounds separately; scale bounds and initial guess; expose callback arg
## Description
- [x] Fixes #75
- [x] Fixes #68 
- [x] Fixes #76 
- [x] Fixes #88

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/SNL-WaterPower/WecOptTool/blob/main/.github/CONTRIBUTING.md)
- [x] The pull request is from an issue branch (not main) on your fork to the [main branch in WecOptTool](https://github.com/SNL-WaterPower/WecOptTool). 
- [x] The authors have given the admins edit access
- [x] All changes adhere to the style guide including PEP8, Docstrings, and Type Hints [... cut]
closed 2022-04-05 by cmichelenstrofer