sandialabs/WecOptTool #80: Verify wave directions when reading BEM data

dtgaebe, 2022-03-11:
**Feature description.**
When doing wec.read_bem() it does not check for wave direction. It should check for wave direction in a similar way like checking for mass

dtgaebe, 2022-03-11:
That would require the class WEC to have wave_direction parameter?!

dtgaebe, 2022-03-11:
Idea is not necessarily to check when reading the bem file, but give the user an easy to understand error message if the BEM file does not contain the wave direction the user specified and if the solver fails to to the different wave directions. 

So no need to add wave_direction parameter to WEC class.

ryancoe, 2022-03-15:
Can this be closed as a result of #81?

dtgaebe, 2022-03-15:
No, not yet. I haven't implemented the error message yet. 
We were thinking about the case that we run the BEM solution for a variety of directions. If the user then only reads the existing BEM solution file, but changed the incident wave direction for the solver. There will be an error, but for the user it might not be trivial to understand what caused the error.

cmichelenstrofer, 2022-03-15:
This check should be in the `solve` method. That method takes in a `wave` object. If the directions in the wave are not included in the BEM then give an error. The other thing we discussed (but which is a bit more complicated to implement) is allowing for the wave to have a subset of the directions in the BEM. Currently for the code to work both the BEM and wave have to have the exact same directions.

dtgaebe, 2022-03-16:
So currently the check is in `wave_excitation()`, which is only called by `solve()`. That's where I continued editing the check so that (for now) it checks if the any of the wave directions is in the BEM solution, which is fine for single directional waves, since they are now checked for a single direction, but I will need to edit the check so that it will test for the entire wave directions being a subset of BEM as you suggest.

cmichelenstrofer, 2022-03-16:
Sounds good, keep it in `wave_excitation`. We also need to implement (not just check) the case where the wave directions are a subset of the directions in the BEM.

cmichelenstrofer, 2022-10-20:
This is in v2.0. Any future improvements will come from the multidirectional wave example in FY23.

closed 2022-10-20 by cmichelenstrofer