sandialabs/WecOptTool #86: Zero frequency component

ryancoe, 2022-03-24:
In computation, we use an impedance where position is the state (flow) variable. This is called `Gi` in our parlance. Thus, we add a zero frequency component equal to the linear stiffness. This done when we create a block matrix.

The zero frequency component is not stored in `wec.hydro`, even though `wec.hydro.Gi` exists.

Our code runs, but is somewhat inconsistent and could probably be improved.

cmichelenstrofer, 2022-03-24:
It is more common to deal with impedance `Zi` (based on velocity) rather than `Gi` (based on position). For `Zi` the zero component is always zero, which is why it is usually ignored.

cmichelenstrofer, 2022-03-24:
`wec.hydro` is created from the output of Capytaine. If we want to add the zero frequency component we will need to add it to everything else in that xarray, e.g. added_mass, Froude-Krylov, radiation damping, etc. `Gi` should absolutely have its zero component, so if this is what we need to do it is worth it. But I am wondering if instead we can run the zero frequency BEM solution too (it should be all zero, so it should be quick, but not sure Capytaine can handle it)

cmichelenstrofer, 2022-03-24:
Capytaine's post-processing function `impedance` returns `Gi` rather than `Zi` which is what is more commonly called "impedance". We should modify Capytaine to output `Zi` and then modify WecOptTool accordingly.

ryancoe, 2022-03-25:
While working on #74, I noticed that there frequency domain output from `wave_excitation` has a zero frequency component.

ryancoe, 2022-08-08:
@cmichelenstrofer realized that including the zero frequency component resulted in lots of divide by zeros, etc. Therefore chose to only show the zero frequency in the results, but not in the BEM and waves. We may need to look at this again later on, as having it exist some places, and not others may be more confusing than its worth.

cmichelenstrofer, 2022-10-20:
This will likely come up again in the future, but for now it is resolved with the mixed approach we agreed for v2.0. Closing for now. I will add a _discussion_ instead.

cmichelenstrofer, 2022-10-20:
Changed my mind. keeping this in Issues but consolidating different ones into this one.

cmichelenstrofer, 2022-10-20:
We have gone back and forth on wether to include the zero-frequency component in the state variables. In theory there is no reason to, they should always be zero. But the argument in favor is that it makes catching user errors easier, and can be more intuitive. We tried making all variables (including wave and BEM results) have zero frequency components, but that did not work out well. Currently (v2.0) we have a mixed approach where some arrays include the zero-frequency component and some do not.

There is also the topic of making the buoyancy/hydrostatics more intuitive. See #97 & #61

- #97
- #61

cmichelenstrofer, 2023-04-15:
I think the implementation we have now is a good approach; I don't see it changing anytime soon. We now also have a tutorial showing how to include buoyancy, gravity, and pretension (Tutorial 2 - AquaHarmonics). Closing this issue.

closed 2023-04-15 by cmichelenstrofer