sandialabs/WecOptTool #87: automatic type checking

ryancoe, 2022-03-24:
I was working on #78, specifically addressing #75... [`scipy.minimize.optimize`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.minimize.html#scipy.optimize.minimize) can handle two types for the `bounds` argument: [`Bounds`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.Bounds.html#scipy.optimize.Bounds) or `sequence`. This is tricky to handle, so I wrote code to handle only `Bounds` types for the `bounds_wec` and `bounds_opt` to `wec.solve`. That got me thinking about checking the types and raising a more helpful error than whatever happens randomly. I was about to do something like

```python
if not isinstance(bounds_wec, Bounds):
    raise TypeError('...')
```

but then I read some more and it looks like there are some sleek solutions for doing this based entirely on the type hints that you put in the function signatures:

 - typeguard: https://github.com/agronholm/typeguard
 - beartype: https://github.com/beartype/beartype _**(from my reading, this looks like the better option)**_ 
 - mypy: http://mypy-lang.org

If I understand correctly, we could use `beartype` by:

1. adding it to our dependencies

2. calling `from beartype import beartype` in each module

3. using the `@beartype` decorator on functions and methods

Any thoughts @cmichelenstrofer?

cmichelenstrofer, 2022-03-24:
That's a philosophical question... 

My preference is what we are doing now: Duck Typing (like Python was intended to be) + type hints (for linters & IDEs) + docstrings (for automatic API documentation). 

But I could go either way. If we're going the static typing way we should do it right (not assert statements). Of the packages you mention above I think by far the most common is `mypy`. I'll have to read about it to see how BearType is different). 

On this topic, it would be good to check if there is (or are plans to) a way to use the tpye hints for the API documentation. Seems redundant to specify the type twice. We are using Sphinx + Numpy docstring format for this right now.

cmichelenstrofer, 2022-03-24:
https://pypi.org/project/sphinx-autodoc-napoleon-typehints/

cmichelenstrofer, 2022-03-24:
@ryancoe putting this link you shared regarding assertions vs exceptions here: https://stefan.sofa-rockers.org/2018/04/16/assertions-and-exceptions/

ryancoe, 2022-06-15:
Verdict: put type-hints in signatures, use [`sphinx-autodoc-napoleon-typehints`](https://pypi.org/project/sphinx-autodoc-napoleon-typehints/) to make them show up in html docstrings

cmichelenstrofer, 2022-10-20:
completed/fixed in v2.0

closed 2022-10-20 by cmichelenstrofer