sandialabs/WecOptTool #97: Intuitive hydrostatics

ryancoe, 2022-04-07:
With #49, we have a conistent way of handling's hydrostatics, buoyancy. However, want to consider:

- is this as intuitive as possible
- can we use capytaine's upcoming hydrostatics module (https://github.com/mancellin/capytaine/pull/106)?

ryancoe, 2022-04-18:
One thing we did not previously appreciate is that **_you do really need to be at equilibrium_** (if you are not, BEM and/or you transfer functions will not be accurate)...

- our previous work (#49) was at least partially premised on allowing for non-zero mean positions
- our whole process of operating on position instead of velocity and having a zero frequency component (#86) was based on this

cmichelenstrofer, 2022-10-17:
v2 uses the new Capytaine hydrostatics. These however are currently changing, e.g. capytaine/capytaine#215, so will need to review in the future.

cmichelenstrofer, 2022-10-20:
Closing this for now. We are using Capytaine's functions in v2.0. I will add the buoyancy topic to the _Discussion_ on zero-frequency component.

closed 2022-10-20 by cmichelenstrofer