sandialabs/WecOptTool #102: Revert #100 once openblas is fixed on conda-forge

cmichelenstrofer, 2022-04-08:
`openblas` `v0.3.20` was added to `conda-forge` 3 days ago. The previous version on `conda-forge` was `v0.3.18`. Reverting back fixes the installation issue in Mac OS. Until this issue is fixed on `conda-forge` we have to install `openblas` explicitly before installing `capytaine`. This is hardcoded into the [CI test](https://github.com/SNL-WaterPower/WecOptTool/blob/main/.github/workflows/python-package.yml). Anyone on Mac trying to install WecOptTool has to do the following:

```python
conda install -c conda-forge openblas=0.3.18
```
before installing Capytaine and WecOptTool

cmichelenstrofer, 2022-04-11:
But I have been thinking that we should release a  Conda package in addition to the PyPI package. This could be done automatically with GitHub Actions like we do with PyPI, so only the initial setup (and eventual updates) will be required.

ryancoe, 2022-04-11:
understood, just would be an easier fix in our README (i.e., wouldn't require macOS-specific instructions)

cmichelenstrofer, 2022-10-20:
fixed in v2.0

closed 2022-10-20 by cmichelenstrofer