sandialabs/WecOptTool #117: use wavespectra formatting

ryancoe, 2022-06-17:
To store information about waves with which to run simulations, we've created our own custom [`xr.DataSet`](https://docs.xarray.dev/en/stable/generated/xarray.Dataset.html) in the [`waves`](https://github.com/SNL-WaterPower/WecOptTool/blob/main/wecopttool/waves.py) module. The [`wavespectra`](https://github.com/wavespectra/wavespectra) project has done something more than this, but along a similar vein. We may want to adopt their structure. One big benefit would be directly using many hindcast and measurement data sources (see I/O formats here: https://wavespectra.readthedocs.io/en/latest/io-input.html#).

```python
In [1]: import matplotlib.pyplot as plt

In [2]: from wavespectra import read_ww3

In [3]: dset = read_ww3("_static/ww3file.nc")

In [4]: dset
Out[4]: 
<xarray.Dataset>
Dimensions:  (dir: 24, time: 9, site: 2, freq: 25)
Coordinates:
  * dir      (dir) float32 270.0 255.0 240.0 225.0 ... 330.0 315.0 300.0 285.0
  * freq     (freq) float32 0.04118 0.0453 0.04983 ... 0.3352 0.3687 0.4056
  * site     (site) int32 1 2
  * time     (time) datetime64[ns] 2014-12-01 2014-12-01T12:00:00 ... 2014-12-05
Data variables:
    dpt      (time, site) float32 dask.array<chunksize=(9, 2), meta=np.ndarray>
    efth     (time, site, freq, dir) float32 dask.array<chunksize=(9, 2, 25, 24), meta=np.ndarray>
    lat      (site) float32 dask.array<chunksize=(2,), meta=np.ndarray>
    lon      (site) float32 dask.array<chunksize=(2,), meta=np.ndarray>
    wspd     (time, site) float32 dask.array<chunksize=(9, 2), meta=np.ndarray>
    wdir     (time, site) float32 dask.array<chunksize=(9, 2), meta=np.ndarray>
```

cmichelenstrofer, 2022-08-01:
I looked into `wavespectra` in detail. I think we should use it but... I don't think we can completely replace having our own structure. We want to be able to handle any type of wave including regular waves and more complex structures like, e.g. the sum of a finite number of regular waves from different directions, with direction not necessarily evenly spaced. To allow for these and irregular (directional or unidirectional) waves, probably the best option is for the `core` to use discrete wave spectra (units of `m^2`) or simply elevation amplitude (units of `m`) indexed by frequencies (evenly spaced) and directions (arbitrary). We should then have functions in `waves` to convert from continuous 1D (units `m^2/Hz`) or 2D (units `m^2/Hz/degree`) to an xarray of discrete elevations. These functions should accept `wavespectra.specArray` and `wavespectra.specDataset`. 

- To convert the continuous 1D to a discrete spectra simply multiply by the fixed frequency spacing Δf. 
- To convert the continuous 2D to a discrete spectra simple multiply by the fixed frequency spacing and fixed direction spacing Δf*Δθ.

cmichelenstrofer, 2022-08-01:
Also, our methods for creating parametric spectra are more general than those in `wavspectra.construct` (only two parametric spectrum functions and a hardcoded spread function, and poor documentation). We should maybe improve `wavespectra.construct` and move our methods there.

ryancoe, 2022-08-03:
@cmichelenstrofer helped me wrap my head around this

[image]

cmichelenstrofer, 2022-08-03:
Long crested uses `S(w) [m^2/Hz]`.

cmichelenstrofer, 2022-10-20:
completed/fixed in v2.0. We will need to help develop `wavespectra` to fix some issues. Will open separate issue.

closed 2022-10-20 by cmichelenstrofer

RubendeBruin, 2022-10-27:
> I looked into `wavespectra` in detail. I think we should use it but... I don't think we can completely replace having our own structure. We want to be able to handle any type of wave including [...]

@cmichelenstrofer , @ryancoe ,

What are your thoughts about: 
https://github.com/wavespectra/wavespectra/pull/40
and
https://github.com/RubendeBruin/wavespectra/blob/bin_to_continuous/docs/spectral_energy_and_bins.rst
in praticular?