sandialabs/WecOptTool #118: autograd -> JAX

ryancoe, 2022-07-01:
Ever since originally adopting [`autograd`](https://github.com/HIPS/autograd), we've been concerned that most of the development energy from [`autograd`](https://github.com/HIPS/autograd) has moved to [`JAX`](https://github.com/google/jax). In addition to continued development, [`JAX`](https://github.com/google/jax) also has more complete functionality (e.g., for `fft`).

We did not use [`JAX`](https://github.com/google/jax) initially because support for MS Windows is not great - users can either compile on their own, use a [third-party binary](https://github.com/cloudhan/jax-windows-builder), or use Windows Subsystem for Linux (WSL) (https://github.com/google/jax#installation).

Given that more direct [`JAX`](https://github.com/google/jax) support for MS Windows does not imminent, we there are two major hurdles preventing us from transitioning:

- **Additional challenge for MS Windows users** - need to see how much of a burden it is to use Windows Subsystem for Linux
- **Changes to our source** - we initially thought this would amount to changing ```import autograd.numpy as np``` to ```import jax.numpy as np```, but apparently there is more to it than that

cmichelenstrofer, 2022-10-20:
Moving this to a discussion. #142

michaelcdevin, 2023-10-10:
@ryancoe @cmichelenstrofer it appears [a Windows-compatible pip install is now available for JAX as of v0.4.13](https://jax.readthedocs.io/en/latest/installation.html#cpu). That's one of the major implementation hurdles out of the way!

cmichelenstrofer, 2023-10-10:
We should try it at some point. But it does require changing the source a bit, so it won't be a small task.

ryancoe, 2023-10-10:
@michaelcdevin - can you quickly see how well the Jax Windows install works as a first step?

michaelcdevin, 2023-10-11:
`pip install jax` works without a hitch on Windows. I tested some of the basic `jax.numpy` and `jax.grad` operations and all seems to work as expected.

michaelcdevin, 2024-06-21:
NumPy v2.0 was released four days ago with various breaking changes. Since `autograd` is no longer maintained, this makes it so **`autograd` is incompatible with current and future releases of NumPy**.

It looks like [`jax` was proactive about maintaining compatibility with NumPy v2.0](https://github.com/google/jax/issues/19246). I added in a NumPy version limitation in 902c17b as a stopgap so WecOptTool doesn't break, but switching from `autograd` to `jax` is a higher priority now so we don't fall behind in NumPy versions.

@cmichelenstrofer @ryancoe

KapilKhanal, 2025-02-15:
I’m exploring JAX for obtaining gradients with respect to BEM data and noticed a few areas where compatibility with Jax could improve:

- [ ] Replace dict/tuple data structures with Pytrees for JAX compatibility (avoids tracing errors).
- [ ] Consider using [xarray_jax](https://github.com/allen-adastra/xarray_jax) to maintain dataset integrity.
- [ ] Use [Equinox](https://docs.kidger.site/equinox/) to preserve class-based structures.

Are these changes or similar updates already planned or implemented maybe in a dev version? Happy to discuss further if needed!