sandialabs/WecOptTool #119: "Irregular frequencies" in BEM coefficiencts --> Interpolate?

dtgaebe, 2022-07-14:
I noticed some numerical instabilities in the BEM coefficients for the WaveBot and frequencies >0.9 Hz. Refining the mesh didn't solve this.
[image]
Those are issues well-known to BEM solvers. We might want to address this in the post-porcessing?

The authors in the following suggest some post-processing techniques and have implemented them in an open-source tool [BEMRosetta ](https://github.com/BEMRosetta/BEMRosetta), written in C++. 
https://www.researchgate.net/publication/355198128_A_post-processing_technique_for_addressing_%27irregular_frequencies%27_and_other_issues_in_the_results_from_BEM_solvers

Me might have some function for our users that detect and interpolate the irregular frequencies as well.

ryancoe, 2022-07-29:
I think that `capytaine` may soon tackle internal frequencies directly (i.e., not a post-processing step; https://github.com/capytaine/capytaine/issues/44), but this paper is useful to know about.

ryancoe, 2023-11-10:
We've heard more recently that `capytaine` is indeed trying to address this problem internally. Thus, for `wecopttool`, we'll continue to consider this problem _out of scope_. However, if there's anyone that needs a more immediate solution, you may find it useful to reference:

- [Kelly et al. 2021](https://www.researchgate.net/publication/355198128_A_post-processing_technique_for_addressing_%27irregular_frequencies%27_and_other_issues_in_the_results_from_BEM_solvers)
- Example code provided here: https://github.com/capytaine/capytaine/issues/44#issuecomment-1714854805

jtgrasb, 2024-12-24:
Closing this issue because [Capytaine has added lids](https://capytaine.github.io/stable/developer_manual/api/capytaine.meshes.collections.html#capytaine.meshes.collections.CollectionOfMeshes.generate_lid) to help with irregular frequency removal and I've added the lids to our tutorials with PR #363.

closed 2024-12-24 by jtgrasb