sandialabs/WecOptTool #122: Tutorial 1 expansion

ryancoe, 2022-08-02:
@cmichelenstrofer and I have discussed how at our current rate we will need ~20 tutorial to cover all of content within WecOptTool. To avoid this, we need to make our tutorials a bit longer and move more quickly. This will be a general effort, but this issue will focus on Tutorial 1.

1. Create geometry, WEC (one DOF WaveBot), and PTO w. constraints
2. Running with and without constraints
3. Plot time history and frequency results (***end of current content***)
4. Optimize hull geometry (as in Tutorial 2) for both electrical and mechanical power
5. Plot optimization results and plot time domain results

^ feel free to edit or comment on the above

michaelcdevin, 2022-08-11:
Following discussions with @cmichelenstrofer, we will organize the first tutorial with the following format to ensure it is easy to follow for new users:

1. Single loop optimization:
* Optimize for mechanical power first
    - Should be accomplished in minimal lines of code
    - Very basic plots to illustrate plotting functionality (probably just power)
* Optimize for electrical power w/ impedance function
    - Constrained + unconstrained
   - Plotting:
     - Include comparison plots for constrained vs unconstrained
     - Position
     - PTO force
     - Power
2. Geometry optimization (basically what is currently Tutorial 2):
 * Optimize for electrical power only
   - Include the same plotting as what is currently in Tutorial 2

cmichelenstrofer, 2022-10-20:
completed/fixed in v2.0

closed 2022-10-20 by cmichelenstrofer