sandialabs/WecOptTool #127: BUG: Scaling of amplitude spectrum in long crested waves

mbruggs, 2022-08-31:
**Describe the bug**
Commit 04c3f0bd6094fa36cb857d7941aaf7d68ddabecf, incorporated via #93 in a10841417c071cfb82ef7c2fcd614d3a8b92d7ec, scaled the amplitude spectrum by a factor 1/(2*pi) for long crested waves. This does not appear to be correct as the resulting wave elevation profile do not match the input Hs and the resulting power output is roughly a factor 6 smaller than the previous commit (497fcc66889c5f9f41d44eeeb536fda5b1e468f9). We have previously run a validation of WecOptTool against other tools and found good agreement for the power output.

**To Reproduce**
1. Install commit 497fcc66889c5f9f41d44eeeb536fda5b1e468f9 of the WecOptTool
2. Run an optimisation which makes use of the long crested wave, eg
   ```python
   Hs = 0.5
   Tp = 10
   fp = 1/Tp
   waves = lcw(f0=f0, nfreq=nfreq, spectrum_func=lambda f: pm(freq=f, fp=fp, hs=Hs))
   ```
3. Note output power and wave elevation
4. Install commit a10841417c071cfb82ef7c2fcd614d3a8b92d7ec of the WecOptTool 
5. Re-run same optimisation and compare wave elevation and power output

**Expected behavior**
The large difference in wave elevation and output power between the two commits is unexpected and does not appear to be correct.

**Observed behavior**
Wave elevation for long-crested PM spectrum (code example above) with
| Hs [m] | Tp [s] | f0 [Hz] | nfreq |
|----|----|----|-------|
| 0.5| 10 | 0.02 | 50 |

Commit 497fcc66889c5f9f41d44eeeb536fda5b1e468f9 (2 runs):
[image]

[image]

Commit a10841417c071cfb82ef7c2fcd614d3a8b92d7ec (2 runs):
[image]

[image]

Qualitatively, the wave elevation runs for a10841417c071cfb82ef7c2fcd614d3a8b92d7ec are too small and not likely to produce a Hs of 0.5m.

**System:**
 - OS: MacOS 12.5.1
 - Python version: 3.9.12
 - WecOptTool version: Locall installed using commits 497fcc66889c5f9f41d44eeeb536fda5b1e468f9 and a10841417c071cfb82ef7c2fcd614d3a8b92d7ec
 - Any other relevant package version: N/A

**Additional information**
The same change also appears to have been made to the `irregular_wave` function but this has not been tested.

Thanks again for all the work on the WecOptTool!

cmichelenstrofer, 2022-10-17:
This should be fixed in V2, but I will keep this issue open until I test it more thoroughly. I am planning on creating some test cases using different types of waves.

mbruggs, 2022-10-19:
Great, thanks very much for looking into it @cmichelenstrofer .

cmichelenstrofer, 2022-10-20:
@mbruggs I haven't had a chance to look at this. But V2.0 is released. If you have time, could you try it again with v2.0 and let me know if the issue is still there? Note that the wave module changed significantly, we can meet one-on-one if you want to setup the example in v2.0.

cmichelenstrofer, 2022-12-01:
With v2.0: 

```python
import wecopttool as wot
import matplotlib.pyplot as plt

f1 = 0.02
nfreq = 50

Hs = 0.5
Tp = 10
fp = 1/Tp

spectrum_func = lambda f: wot.waves.pierson_moskowitz_spectrum(f, fp, Hs)
spectrum_name = "Pierson Moskowitz"

efth = wot.waves.omnidirectional_spectrum(f1, nfreq, spectrum_func, spectrum_name)
waves = wot.waves.long_crested_wave(efth)

waves_fd = wot.complex_to_real(waves.values, False)
waves_td = wot.time_mat(f1, nfreq, 1, False) @ waves_fd

time = wot.time(f1, nfreq)

plt.plot(time, waves_td)
```

[image]

Similar results to those after #93. 

However, I think these results are the correct ones. Calculating the significant wave height manually in the time domain

```python
import numpy as np
4 * np.std(waves_td)
```
gives `0.5005591213407394 `. 

And in the frequency domain:
```python
m0 = np.sum(np.abs(waves.values))*f1
4 * np.sqrt(m0)
```
gives `0.44028708773480496` (not sure why less accurate than in the time-domain, would have expected the opposite). 

@ryancoe any thoughts on this?

cmichelenstrofer, 2022-12-01:
@mbruggs @ryancoe qualitatively I would expect the waves to follow the historical definition of "the average of the 1/3 largest waves". And even though this is only 50 seconds, it does look unlikely.

cmichelenstrofer, 2022-12-01:
OK There is a bug. Hold on.

cmichelenstrofer, 2022-12-02:
Created a PR #167 that should fix this in v2.0. @mbruggs we (@dtgaebe or myself) can work with you to move to v2.0. It would be great if you could then redo your validation tests.

cmichelenstrofer, 2022-12-13:
OK, so I was confused. There was no bug in V2.0. But I used that PR to add some tests and cleanup for the irregular waves.

commit 41437ef6 (2022-12-14): BUGFIX: Factor of sqrt(2) in irregular_wave, closes #127 (#167)

 * Improves testing for irregular waves
 * Fixes bug in irregular_wave (missing sqrt(2) factor)
 * Arg validation for fd_to_td
closed 2022-12-13 by cmichelenstrofer