sandialabs/WecOptTool #129: BUG: A recently updated capytaine 1.4.1 shows inertia and hydrostatic error with tutorial 1 and 2

Jinhyuk-Yu, 2022-09-21:
Description of the bug
I started using WecOptTool recently, and I ran tutorial 1 wavebot.ipynb and tutorial 2 wavebot optimization.ipynb. But in capytaine 1.4.1, WecOptTool showed the inertia and hydrostatic calculation error. 

The error message is below:
[image]

I reinstalled capytaine with 1.3 version and WecOptTool worked, but in 1.4 it did not work.
I would like to ask how to fix this problem

**System:**
 - OS: Anaconda for Window 10
 - Python version: 3.9.12
 - WecOptTool version: Most recent version (installed Sep 14th, 2022)
 - Any other relevant package version: Capytaine 1.4.1

rebeccamccabe, 2022-09-26:
I encountered this error too (also on Capytaine 1.4.1), and I am including the full traceback: the bug occurs when running the bem, and is ultimately from the `rao_transfer_function` method in capytaine's `impedance.py`.
[image]

ryancoe, 2022-09-26:
Thanks @rebeccamccabe and @Jinhyuk-Yu! This is a mismatch of versions between WecOptTool and Capytaine. We have this fixed in the soon-to-be-released v2 of WecOptTool (https://github.com/SNL-WaterPower/WecOptTool/tree/v2.0.0-dev), but in the meantime you can simply use Capytaine v1.3. We'll add a pull request to correct this in our documentation in a few minutes.

ryancoe, 2022-09-26:
After a bit of a run-around, we've observed some complex interactions between operation systems, capytaine versions, and maybe openblas (see #100 and #102). We will ensure that this problem in fully-resolved in v2, but in the meantime you should be able simply use conda v1.3

```
pip install git+https://github.com/LHEEA/meshmagick.git@3.3
conda install -c conda-forge capytaine=1.3
pip install wecopttool
```

Jinhyuk-Yu, 2022-10-03:
Thank you for your contribution and checking this error. 
I am using WecOptTool with the previous capytaine version 1.3 and it works well. 
Hopefully, the next version will cover this problem well.

cmichelenstrofer, 2022-10-20:
Resolved in v2.0. It shouldn't happen, but let us know if you still get version errors. I've checked: Python 3.9 and 3.10, and Capytaine 1.4.2.

closed 2022-10-20 by cmichelenstrofer