sandialabs/WecOptTool #153: Pip installation on Windows broken due to Capytaine.

cmichelenstrofer, 2022-10-31:
We can now install all required packages simply using `pip install wecopttool` (need Fortran compiler) or `conda install -c conda-forge wecopttool`. The one exception is *Windows* using `pip`, which doesn't work due to extra steps needed to compile `capytaine`. See capytaine/capytaine#115.

ryancoe, 2022-12-15:
I think this is fixed now, right @cmichelenstrofer ?

cmichelenstrofer, 2022-12-15:
@ryancoe I don't think so. This is not on our end, it is a Capytaine issue. You cannot do `pip install capytaine` on Windows. You have to either use `conda` or follow a bunch of extra steps. So installing WecOptTool on Windows requires that you first install Capytaine using Conda.

mancellin, 2023-06-21:
There is now a pip package for Capytaine on Windows :tada:

ryancoe, 2023-10-06:
re-assigning this to @michaelcdevin as he's a Windows users ;)

michaelcdevin, 2023-10-11:
Can confirm, `pip install wecopttool` installs Capytaine via pip and is fully functional on Windows: https://github.com/michaelcdevin/WecOptTool/actions/runs/6485178053/job/17610688948 (the CI run is failing due to an unrelated issue).

I think we can safely remove [this disclaimer](https://github.com/sandialabs/WecOptTool/blob/e6cda25fd4fed39e2354237b75417c7f9fee03be/README.md?plain=1#L28) in the README now. Thanks @mancellin for updating this!

closed 2023-10-13 by michaelcdevin