sandialabs/WecOptTool #162: BUG: Installing via Conda hangs

michaelcdevin, 2022-11-17:
**Describe the bug**
Following the instructions in the README for the Conda installation fails, as the installation fails on the "solving environment" step

**To Reproduce**
Steps to reproduce the behavior:
Run `conda install -c conda-forge wecopttool` in terminal.

**Expected behavior**
WecOptTool installs from conda-forge with all needed dependencies except `gmsh` and `pygmsh`.

**Observed behavior**
Terminal output:
```
Collecting package metadata (current_repodata.json): ...working... done
Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
Solving environment: ...working... failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... 
```
Install hangs indefinitely on the last step.

**System:**
 - OS: Windows 10
 - Python version: 3.9.12
 - WecOptTool version: v2.0.1a1

**Additional information:**
Behavior has also been replicated on Mac.

michaelcdevin, 2022-11-17:
Downloading the package [directly from the Anaconda website](https://anaconda.org/conda-forge/wecopttool/files) and running `conda install wecopttool-2.0.1a1-pyhd8ed1ab_0.tar.bz2` works without issue. This is the same file as when installed from PyPI, so no surprise here, but just wanted to confirm that WecOptTool itself is in good shape -- must be something to do with the dependencies in the Conda build.

michaelcdevin, 2022-11-22:
Come to find out that the conda build doesn't hang _indefinitely_... just a very, very long time. After about an hour on my machine, the "solving environment" step finally moves forward:

```
Collecting package metadata (current_repodata.json): ...working... done
Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
Solving environment: ...working... failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
Solving environment: ...working... 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed
```

The "Found conflicts!" step lasts >30 minutes in itself, and has a very verbose error stream ([here](https://github.com/SNL-WaterPower/WecOptTool/files/10070480/install_output_errors.txt)) and I'm still working through this. But clearly there's various dependency issues that didn't occur when doing the local build install when testing, nor in the conda-forge CI.

michaelcdevin, 2022-11-22:
Also worth noting, I was able to manually build the conda environment with all the needed dependencies, then pip installed wecopttool, gmsh, and pygmsh, and all the unit tests ran fine. However, the Jupyter package takes a very long time (20 minutes on my laptop) to solve its environment in itself, so I suspect the issue may lie there.

ryancoe, 2022-11-22:
Not to rush this, but as we work through this issue we may want to add an **Option 3** to the installation instructions our `README` that uses something like the following

```bash
pip install git+https://github.com/LHEEA/meshmagick.git@3.3
conda install -c conda-forge capytaine
pip install wecopttool
```

cmichelenstrofer, 2022-11-22:
we need pygmesh not meshmagick.  It would be something like: `conda install -c conda-forge pygmesh`.

cmichelenstrofer, 2022-11-22:
It would be: 
```python
pip install gmsh pygmsh
conda install -c conda-forge capytaine wavespectra
pip install wecopttool
```

cmichelenstrofer, 2022-11-22:
Changed option 1 to the solution above, which is what is working right now

cmichelenstrofer, 2022-11-22:
also updated option 2 since currently gmsh and pygmsh need to be installed separately even with pip

michaelcdevin, 2022-11-23:
I've tried changing the conda `channel_priority` to `flexible`, `strict`, and `disabled`, all without success.

Interestingly, `conda install -c conda-forge wecopttool` works without issue on Red Hat Linux, even with a `channel_priority` of `strict`, so it looks like it might be OS-related. All the tests not involving gmsh also pass (the pip install of gmsh doesn't entirely work on Linux as the OpenCASCADE kernel [needs to be compiled from source](https://gitlab.onelab.info/gmsh/gmsh/-/issues/1808), but that's irrelevant to the issue on hand since all the conda dependencies clearly installed correctly).

closed 2022-12-16 by michaelcdevin

michaelcdevin, 2022-12-17:
This issue has finally been resolved. The short version of the solution is that `numpy >=1.20` had to be removed from `requirements/host` in our [`meta.yaml`](https://github.com/conda-forge/wecopttool-feedstock/blob/main/recipe/meta.yaml) file, as was done in conda-forge/wecopttool-feedstock@a001942. I forgot the original reason why I even included `numpy` in `requirements/host`, but its removal will not affect `wecopttool` regardless.

The long version is a convoluted versioning issue that I'm documenting below for future reference (likely for my future self).

michaelcdevin, 2022-12-17:
**Long solution explanation:** Besides the `meta.yaml` file, conda recipes can optionally include a `conda_build_config.yaml` file as well. To my understanding, `conda_build_config.yaml` is needed if the conda package uses [pinned dependencies](https://conda-forge.org/docs/maintainer/pinning_deps.html) or [has several variants for compiling binaries in different environments](https://conda.io/projects/conda-build/en/latest/resources/variants.html). WecOptTool doesn't do either of these, and the conda-forge documentation does not otherwise reference `conda_build_config.yaml` for [contributing packages](https://conda-forge.org/docs/maintainer/adding_pkgs.html#) or [maintaining packages](https://conda-forge.org/docs/maintainer/updating_pkgs.html), so I did not add it.

However, what is undocumented is that `conda-build` still uses default values if versions are not specified in `conda_build_config.yaml`, which resulted in the following warning to appear when building WecOptTool locally using `conda build .`:
```
No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.16
WARNING:conda_build.metadata:No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.16
Adding in variants from internal_defaults
```
After some research, I figured this was not a concern here, as none of our dependencies need to build explicitly against `numpy` (which requires [special linking separate from the `conda_build_config.yaml` file](https://conda-forge.org/docs/maintainer/knowledge_base.html#linking-numpy) anyway), so this was ignored. However, what I missed is that including `numpy <=1.20` in the `requirements/build` section of `meta.yaml` adds `numpy` to the build process when solving for the required environment, [which occurs blind of any designations made in the `conda_build_config.yaml` file](https://github.com/conda/conda-build/issues/3170#issuecomment-439067240). So in the end, `conda_build_config.yaml` initially specifies `numpy == 1.16` when initializing the build, and later `numpy <=1.20` is specified while parsing `meta.yaml`, which causes incompatibilities in any package using `numpy` (i.e. most of them) and thus the hanging issue when running `conda install wecopttool`.

I believe the reason the conda install worked on Linux was because conda-forge hosts on Linux by default, and `conda_build_config.yaml` is only called when building on a platform different than the host platform. Therefore, the `numpy` version discrepancy never occurs since `conda_build_config.yaml` is never called.