sandialabs/WecOptTool #165: Alternative geometry package

cmichelenstrofer, 2022-12-01:
In our example we currently use `pygmsh` for our examples. However this package is not in `conda-forge`, see meshpro/pygmsh#558. Now that we are using `conda` packaging in addition to `pip` we won't be able to include `pygmsh` with `wecopttool`. Instead we would have to ask users to install it as a separate step. 

We considered `pygalmesh` but it is not very user friendly and lacks documentation. We should explore other packages that are in both `PyPi` and `conda-forge`. @ryancoe  pointed out [PyVista](https://docs.pyvista.org/index.html).

michaelcdevin, 2022-12-01:
PyVista looks very promising. Its UI is simple and has terrific documentation. I will test it out and see if it will work for our use case well, i.e. see what it takes to create the [WaveBot mesh](https://github.com/SNL-WaterPower/WecOptTool/blob/main/wecopttool/geom.py#L67-L91) with it.

ryancoe, 2022-12-01:
I agree `PyVista` looks polished and has a lot of development energy. My only hesitation is that the main emphasis seems to be visualization, with geometry/mesh creation being somewhat of a tangent. Maybe I'm wrong about this. Or maybe those capabilities are still good. I'm under the impression that [`gmsh`](https://gmsh.info/) is a pretty capable ("pro level") meshing tool.

michaelcdevin, 2022-12-01:
They do seem to be leveraging themselves as an alternative to VTK focused on visualization, but their mesh creation capabilities still seem pretty robust. I think `gmsh` is still *more* robust, but PyVista is able to generate meshes from combinations of geometric objects and from structured or unstructured grids, which should cover all our bases I think.

My only concern is I don't see any ability to change the fineness/coarseness of a mesh (a la `mesh_size_factor` from `gmsh`). I've only taken a cursory look through the API documentation though, and I would be surprised if this were not an option.

michaelcdevin, 2023-01-17:
After quite a bit of investigation into PyVista, I think we should find another meshing package. I was able to recreate the WaveBot and AquaHarmonics bodies from `geom.py` and the UI is straightforward, but the mesh boolean operations are very buggy, especially when altering the resolution of the mesh (e.g. generates meshes with open boundaries, generates closed meshes but has very poor paneling or meshes with very rough paneling, sometimes just silently crashes without explanation). I think @ryancoe's assessment was accurate in that PyVista is a visualization tool first and a mesh creation tool second.

michaelcdevin, 2023-01-17:
I also looked into another meshing package called [`trimesh`](https://github.com/mikedh/trimesh), which was very easy to use, creates very clean meshes, and is even faster than `pygmsh`. However, unlike `pygmsh`, `trimesh` does not use [Delaunay triangulation](https://en.wikipedia.org/wiki/Delaunay_triangulation), which means that the panels of the mesh are extremely long and thin, which isn't good for use in BEM since the height is hardly being discretized at all. I cannot find a function that changes the triangle formulation used in `trimesh`, so finding a Conda-based package that meshes with Delaunay triangulation with a simple UI is still under investigation. [`MeshPy`](https://github.com/inducer/meshpy) looks promising (and has an [up-to-date conda-forge installation](https://anaconda.org/conda-forge/meshpy), despite not being mentioned in the install documentation), so that might be looking into next.
[image]

cmichelenstrofer, 2023-01-17:
I propose **not** looking for an alternative, since none of these big packages seem to work perfectly. The geometry generation is not really needed in WecOptTool so we can leave it up to the users. But to use our examples and tutorials they will need to download `pygmsh` through pip (as well as other packages, e.g. Matplotlib, etc). We already provide an easy way to download all these through the use of *optional requirements*.  This is described in the README as a required step to run the tutorials: 

```python
pip install wecopttool[geometry] jupyter
```

michaelcdevin, 2023-01-17:
I'm okay with that. It would be nice to have everything conda-installable, but since the geometry generation isn't really a core feature of WecOptTool, you're right that it's probably not worth the effort to find an alternative at the moment. And if meshpro/pygmsh#558 gets resolved at some point that will address that issue anyway.

This will leave #179 unresolved, but there are easy alternate fixes for that, so we can still close this for now.

closed 2023-01-17 by michaelcdevin

cmichelenstrofer, 2023-01-17:
> This will leave #179 unresolved, but there are easy alternate fixes for that, so we can still close this for now.

Oh good point, I had forgotten about this.