sandialabs/WecOptTool #168: Documentation improvements: meshing, frequencies

ryancoe, 2022-12-08:
We've discovered/remembered a couple important things that are worth noting in the documentation. We'll put together a list here and then incorporate into the documentation.

- When choosing the number of frequencies, you should consider
  - If the response will be nonlinear (e.g., constraints exist, nonlinear kinematics), you will need higher frequency components 
  - Must always have 2x the incoming wave frequency, as the power absorption will take place at this frequency

dtgaebe, 2022-12-08:
- When choosing the number of frequencies, you should consider 
    - The expected behavior of the hydrodynamic coefficients for high frequencies
        - `added_mass[-n:-1]` → a constant value
        - `radiation_damping[-n:-1]` → zero
    - Limiting the occurrence of irregular frequencies in the hydrodynamic coefficients (combination between ` mesh_size_factor` and frequencies)
- When choosing the  ` mesh_size_factor`, you should consider 
    - Limiting the occurrence of irregular frequencies in the hydrodynamic coefficients (combination between ` mesh_size_factor` and frequencies)
    - Double checking that the hydro-statics ( mass and stiffness) match the values you'd expect

cmichelenstrofer, 2023-04-15:
As we discussed in our last meeting, we do not need to capture the 2x frequency for power. It would not have an effect on the solution, only on post-processing: FFT of power. This might be worth exploring/testing though.

PR #169 (2023-05-21): DOCUMENTATION: frequencies, meshing, copyright
Closes #168
Closes #195 
Closes #184
closed 2023-05-21 by michaelcdevin