sandialabs/WecOptTool #171: Use new features in Capytaine 1.5

cmichelenstrofer, 2022-12-15:
See the [changelog](https://github.com/capytaine/capytaine/blob/master/docs/changelog.rst#new-in-version-15-2022-12-13) for a full list of new features.
In particular:

- New predifned mesh functions. Old geometric bodies functions deprecated. We use these in our tests (to create a test square barge). 
- Improved methods for hydrostatic stiffness and inertia, values saved as part of the floating body.

cmichelenstrofer, 2023-03-07:
the geometric bodies are considered deprecated but still work. Ideally we stop using them soon.

jtgrasb, 2023-10-17:
@cmichelenstrofer I've been looking into saving the hydrostatics as part of the floating body. This can be done pretty easily by defining center of mass and rotation center, then using the compute_hydrostatics() function before running bem, and I plan to make a PR to amend this. One question - should I just get rid of the hydrostatics.py script? The only part of it that seems  necessary at this point is the set_property function which could be moved to the geom.py or core.py scripts. Do you think its reasonable to remove the hydrostatics file?

cmichelenstrofer, 2023-10-17:
Yes, I would agree with that; let's get rid of `hydrostatics.py`. I assume the `_set_property` would be called in `core`, right? In that case I say we move it there. This means we do not need to ask the user for `mass` or `hydrostatic_stiffness` when creating the WEC or solving. We assume it is in the floating body and if not use a default. Correct? I would make sure that these options work well with multi-degree of freedom and multiple bodies...

jtgrasb, 2023-10-17:
@cmichelenstrofer Yeah, it would be called from the `run_bem()` function in `core`, so I'll move it there.

That's correct. My thought is to still leave the variables as optional for the user to add instead of the bem data if desired. There's also an error set up in core that doesn't allow you to override mass or stiffness from BEM with a custom value. I plan to get rid of this error and add a warning that just indicates that the BEM data was overwritten.

Agreed, will make sure to test this out.

cmichelenstrofer, 2023-10-17:
@jtgrasb OK, your call on whether to still allow for optional mass & hydrostatics inputs. I would lean towards not allowing it... this would reduce the number of inputs and the things we need to maintain (mass and hydrostatics are now capytaine's problem and wecopttool becomes smaller. I am hoping to do similar with waves and `wavespectra`.) @ryancoe thoughts?

cmichelenstrofer, 2023-10-17:
I looked at the changelog for v2 and the only other change I think we could benefit is:

> Add [:meth:`~capytaine.bodies.bodies.FloatingBody.minimal_computable_wavelength`](https://github.com/capytaine/capytaine/blob/master/docs/changelog.rst#id118) to estimate the wavelengths computable with the mesh resolution ([:pull:`341`](https://github.com/capytaine/capytaine/blob/master/docs/changelog.rst#id120)).`

We can use this function to determine the largest frequency to use in the frequency vector to aoid the "insufficient mesh resolution" warning. It would be good practice both for us to use and to show users how to use. 

@ryancoe @dtgaebe FYI

dtgaebe, 2023-10-17:
> @jtgrasb OK, your call on whether to still allow for optional mass & hydrostatics inputs. I would lean towards not allowing it... this would reduce the number of inputs and the things we need to [...]

@cmichelenstrofer, @jtgrasb I would allow to override it. There are cases where the dry-mass does not equal the displaced volume, which I think capytaine calculates?

closed 2023-11-01 by cmichelenstrofer