sandialabs/WecOptTool #175: BUG: `pip` install fails because of new NumPy dependency

cmichelenstrofer, 2022-12-22:
* Cannot be fixed from our repo since it is one of our dependencies (capytaine) that now fails to build.
* HOLD V2.1 release until this is fixed.
* See capytaine/capytaine#273 for status

cmichelenstrofer, 2023-01-03:
1.24.1 was [supposed to fix it](https://github.com/numpy/numpy/issues/22871#issuecomment-1364754628) but I just checked and it did not...

michaelcdevin, 2023-01-03:
Adding `charset-normalizer` to the pip install during the build documentation seems to fix it (see https://github.com/numpy/numpy/issues/22871#issuecomment-1367426261). Hopefully this gets wrapped more cleanly into Capytaine and/or NumPy at some point soon, but this is a workable solution for our current purposes.

michaelcdevin, 2023-01-03:
Added in [f5b9e68](https://github.com/SNL-WaterPower/WecOptTool/commit/f5b9e68964979c4e028986c96ac384857b4951c6)

cmichelenstrofer, 2023-01-03:
let's keep this issue open until numpy/numpy#22871 is resolved and hopefully we can remove `charset-normalizer`.

michaelcdevin, 2023-01-17:
capytaine/capytaine#273 is now closed -- it looks like the usage of `charset-normalizer` is the selected NumPy solution. The original CI failure in WecOptTool has been resolved with this solution. The CI still fails to build the website documentation, but this will be tracked in #182

closed 2023-01-17 by michaelcdevin