sandialabs/WecOptTool #176: Non-linear PTO losses

cmichelenstrofer, 2022-12-22:
Moving discussion from #173 here.

The description of the electric power as 

$P_e = P_m * eff = P_m * (1-loss)$ 

cannot correctly account for cases in the (torque-velocity) axes. Instead we should use 

$P_e = P_m + P_{loss}$,

 where $P_{loss}$ is a user-supplied function of torqque and shaft (rotational) velocity. 

For simple (low velocity) cases this model should work:

$P_{loss} = (T/k_T)^2 R_w$,

where $T$, $k_T$, and $R_w$ are the input torque, the motor's torque constant, and the motor's winding resistance, respectively. 

For more complex cases multiple mechanisms contribute to the electric power loss. 
On possible approach is determining this function for your motor and fitting a parametric function to it, such as a polynomial. 
This would allow for it to be differentiable and usable in WecOptTool.

cmichelenstrofer, 2022-12-28:
@ryancoe changes look good! Why "(mostly)" non-linear?

dtgaebe, 2023-01-19:
An approach for the suggested polynomial fitting is described in [[Mahmoudi2017](https://ieeexplore.ieee.org/author/38574727200)].
Assume you have a **motor** efficiency $\eta^{mtr} (\tau, \dot{\theta})$, thus 
```math
P_m^{mtr}  = P_e^{mtr}  \cdot \eta^{mtr} (\tau, \dot{\theta}).
``` 
With the current implementation $P_e = P_m + P_{loss}$, we can convert the **motor** efficiency into an equivalent loss map
```math
P_{loss}^{mtr} = P_e^{mtr}  - P_m^{mtr}  = \frac{1}{\eta^{mtr}}P_m^{mtr}  - P_m^{mtr}  = P_m^{mtr}  \left( \frac{1-\eta^{mtr}}{\eta^{mtr}} \right)
```
Now, [[Mahmoudi2017](https://ieeexplore.ieee.org/author/38574727200)] suggest fitting the motor losses with physics based polynomial of $\tau$ and $\dot{\theta}$.
The parametric function $P_{loss}^{mtr,fit}(\tau, \dot{\theta})$ is very likely non-zero at the axes, thus extrapolates copper losses (such as $I^2R$) towards the axes, when $P_m \rightarrow 0^+$.

If we now assume symmetric power loss for generating and motor operation we can approximate the loss map over all quadrants
```math
P_{loss} = P_{loss}^{mtr,fit}(|\tau|, |\dot{\theta}|).
```

closed 2023-01-19 by dtgaebe