sandialabs/WecOptTool #180: Feature request: function to evalute PTO effort and flow variables with states

dtgaebe, 2023-01-12:
**Feature description.**
The user might be interested in evaluating the PTO effort and flow variables after the PTO impedance. For example those could be electric current and voltage.

**Describe the solution you'd like**

Ideally, the user could use the solution and call a function in `pto.py`  analogous to `pto.velocity(wec, x_wec, x_opt, waves)` to receive the effort and flow variables after the `pto_impedance` and therefore after `pto.transfer_mat`. The physical meaning of those variables would depend on the type of `pto_impedance`.

The desired effort and flow variables need both `pto.velocity(wec, x_wec, x_opt, waves)` and `pto.force(wec, x_wec, x_opt, waves)`. It seems most suitable to compute them in a single function with two outputs, instead of the computation in `pto.power`.

**Describe alternatives you've considered**
Asking to user to post-process this on their own cannot be a solution.

**Interest in leading this feature development?**
I can lead this.

cmichelenstrofer, 2023-01-13:
The option that would make it the most consistent with the rest of the PTO class is two different functions, one for effort and for flow. But the way it is implemented right now, that would require doing the same exact matrix multiplication in both and ignoring half the solution.

cmichelenstrofer, 2023-04-15:
@dtgaebe can you confirm this is resolved/implemented in main, and close this issue?

dtgaebe, 2023-04-17:
This has been introduced with [v2.2.0](https://github.com/SNL-WaterPower/WecOptTool/tree/v2.2.0 ).

closed 2023-04-17 by dtgaebe