sandialabs/WecOptTool #181: remove sine component of 2-point wave.

cmichelenstrofer, 2023-01-13:
Our states `x_wec` (and `x_opt` for the unstructured controller) consist of the Fourier coefficients of the WEC position (and PTO force). We include the mean (DC) component plus the cosine and sine components for each frequency. However, the sine component for the largest frequency has no effect since that sine is evaluated at multiples of pi and is therefor always zero. 

See [this explanation](https://www.oceanopticsbook.info/view/surfaces/level-2/wave-representations) for example:
> The smallest wave that can be captured correctly in a sampling scheme has a period of twice the sampling interval. This called the two-point wave or two-point oscillation. Note that a sine term with period would be sampled at arguments of where the sine is zero and is thus not detectable.

cmichelenstrofer, 2023-02-10:
Some clarification from our discussion yesterday:

- We **are** using that last sine component because our constraints use `nsubsteps>1` and we use a non-square time matrix. The last sine component does not affect the dynamics (equality constraint) or the objective function, but affects other constraints. 
- An alternative approach is to use a larger number of frequencies and no `nsubsteps>1`. We would then use the FFT which is efficient for large problems. But there is still a question about whether autograd handles the FFT correctly.

Actions:

- [ ] Run the tutorial with 4x more frequencies in the same range (0.25x `f0`) and verify we get similar results. 
- [ ] Verify that autograd can handle the FFT
- [ ] Consider moving to FFT instead of time-matrix and not allowing `nsubsteps`

For now, though, let's remove the sine component.

michaelcdevin, 2023-02-10:
> Some clarification from our discussion yesterday:
>
>     * We **are** using that last sine component because our constraints use `nsubsteps>1` and we use a non-square time matrix. The last sine [...]

These new actions will be tracked in #194.

PR #191 (2023-02-16): Remove sine component of 2-point wave
## Description
As the imaginary component of the highest sampled frequency will always evaluate to zero, it does not provide any meaningful gradient information to the optimization algorithm, and thus slows down the solve step unnecessarily. This removes this component from WecOptTool entirely (reducing the number of states by 1) to improve performance, and updates the docstrings, tutorials, and unit testing suite to reflect this. Resolves #181.

## Type of PR
- [x] New feature

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://g [... cut]
closed 2023-02-16 by cmichelenstrofer