sandialabs/WecOptTool #182: BUG: Documentation site build fails due to Capytaine symbol not found

michaelcdevin, 2023-01-17:
On a new release, the site documentation build encounters the following exception:
```
Configuration error:
There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.10.9/x64/lib/python3.10/site-packages/sphinx/config.py", line 350, in eval_config_file
    exec(code, namespace)
  File "/home/runner/work/WecOptTool/WecOptTool/docs/source/conf.py", line 7, in <module>
    from wecopttool import __version__, __version_info__
  File "/opt/hostedtoolcache/Python/3.10.9/x64/lib/python3.10/site-packages/wecopttool/__init__.py", line 32, in <module>
    from wecopttool.core import *
  File "/opt/hostedtoolcache/Python/3.10.9/x64/lib/python3.10/site-packages/wecopttool/core.py", line 68, in <module>
    import capytaine as cpy
  File "/opt/hostedtoolcache/Python/3.10.9/x64/lib/python3.10/site-packages/capytaine/__init__.py", line 21, in <module>
    from capytaine.bem.solver import Nemoh, BEMSolver
  File "/opt/hostedtoolcache/Python/3.10.9/x64/lib/python3.10/site-packages/capytaine/bem/solver.py", line 24, in <module>
    from capytaine.green_functions.delhommeau import Delhommeau
  File "/opt/hostedtoolcache/Python/3.10.9/x64/lib/python3.10/site-packages/capytaine/green_functions/delhommeau.py", line 15, in <module>
    import capytaine.green_functions.Delhommeau_f90 as Delhommeau_f90
ImportError: /opt/hostedtoolcache/Python/3.10.9/x64/lib/python3.10/site-packages/capytaine/green_functions/Delhommeau_f90.cpython-310-x86_64-linux-gnu.so: undefined symbol: f2py_size
```

Likely related to #175 and capytaine/capytaine#273. This does not appear to be a CI caching issue, as WecOptTool uses `capytaine ==1.4.2`. Testing with `capytaine ==1.5` yields the same exception, except `Delhommeau_f90` is replaced with `Delhommeau_float64` in the `ImportError`.

The CI does not technically fail when this occurs, but the website still fails to build.

michaelcdevin, 2023-01-18:
Did a bit more troubleshooting today without success. The following attempted solutions still results in the same error:
* Adding `pip3 install charset-normalizer` to the "Install WecOptTool for documentation" and/or the "Build documentation" release workflows
* Adding `pip3 install numpy` to the "Build documentation" release workflow
* Adding `from numpy import f2py` to `__init__.py`
* Incrementing the `CACHE_NUMBER` in the push workflow

I suspect this may be on Capytaine's side, so see capytaine/capytaine#279.

@cmichelenstrofer is there anything preventing us from using the Conda version of Capytaine via the environment YAML file (as done in [`push.yml`](https://github.com/SNL-WaterPower/WecOptTool/blob/9a49da056de3ea89014b4bdf9ef0953199fdefe4/.github/workflows/push.yml#L134-L135))  for the website build? Since this error is with the Fortran binaries in Capytaine, that could be a workaround.

cmichelenstrofer, 2023-01-18:
I couldn't figure out how to get the tutorials to run correctly (Jupyter use the correct conda kernel). It is probably doable, I did not spend that much time on it. Up to you if you want to go down that road. The tutorials are run through part of the sphynx documentation build process.

michaelcdevin, 2023-01-18:
I will hold off to see if there's an easy fix from the Capytaine side. If not, I'll look into figuring out the tutorials issue when using Conda.

michaelcdevin, 2023-01-27:
I've identified the issue (sorta): removing `wavespectra` as a package dependency fixes the issue and allows Capytaine to import successfully. Note this only works since `wavespectra` is only used for the testing suite, so not including it explicitly for the website build works okay (but the actual fix for this should be more graceful).

I'm not entirely sure what exactly in the `wavespectra` install is causing the issue. My initial guess was that `wavespectra` is dependent on [`numba`](https://github.com/numba/numba), which currently requires `numpy<1.24`. So when doing the `pip install .[dev, geometry]` step (before I removed `wavespectra`), `numpy=1.24.1` is installed as a WecOptTool direct dependency, then `numpy=1.23.5` is installed for `numba`, which prompts the uninstall of version 1.24.1. However, I tried
```
pip3 install 'numpy<1.24' charset-normalizer
pip3 install capytaine
python -c 'import capytaine'
```
and that ran without issue, so there must be something I'm missing.

michaelcdevin, 2023-01-28:
Issue has been fixed: it was the `numpy` version issue combined with a few other things. See https://github.com/capytaine/capytaine/issues/279#issuecomment-1407478394 for details.

PR #187 (2023-02-16): Fix documentation website deployment
## Description
Allows WecOptTool website deployment to work again following changes in NumPy v1.24.0. Fixes #182.

## Type of PR
- [x] Bug fix
- [ ] New feature
- [ ] Documentation
- [ ] Other: (specify)

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/SNL-WaterPower/WecOptTool/blob/main/.github/CONTRIBUTING.md)
- [x] The pull request is from an issue branch (not main) on *your* fork, to the [main branch in WecOptTool](https://github.com/SNL-WaterPower/WecOptTool).
- [x] The authors have given the admins edit access
- [... cut]
closed 2023-02-16 by cmichelenstrofer