sandialabs/WecOptTool #199: Feature request: Nonlinear hydrodynamics

michaelcdevin, 2023-02-16:
**Feature description.**
Incorporating the nonlinear effects of hydrostatic buoyancy and Froude-Krylov forces based on instantaneous wave elevation and WEC position, [such as in WEC-Sim](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#nonlinear-buoyancy-and-froude-krylov-excitation).

**Issue addressed.**
Improves the accuracy of the dynamics equation, as these nonlinear hydrodynamic effects can be significant for WEC design.

**Describe the solution you'd like**
Be able to evaluate the WEC at an instantaneous position and determine what elements are submerged at that moment. Incorporate wave stretching for panels above MWL. This is certainly doable, but making this computationally efficient could be a big hurdle.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered. Also explain any current workarounds.

**Interest in leading this feature development?**
I am open to leading this as time allows.

cmichelenstrofer, 2023-04-15:
The code should be able to handle this as an additional force. I suggest we start by creating an example based on tutorial 1 and add the non-linear forces as additional forces. We can then think about how to make it more formally a part of the code.

ryancoe, 2023-10-27:
@cmichelenstrofer - In the near term, I think it might be best to do this as an ad-hoc function in an example, rather than trying to handle it completely generally in the source. After that, we could potentially move towards a general solution. Any thoughts?

cmichelenstrofer, 2023-10-30:
that sounds good.