sandialabs/WecOptTool #213: BUG: Scaling unconstrained problem generates inconsistent WEC kinematics

michaelcdevin, 2023-03-29:
**Describe the bug**
When using the `WEC.solve()` routine to optimize for average electrical power for an unconstrained WEC with scaling parameters (`scale_x_wec`, `scale_x_opt`, `scale_obj`) ≠ 1, the solution converges to a consistent results, but the WEC kinematics are unexpected and inconsistent. For example, a given WEC may always converge to -10 W optimal average power, but the time domain WEC response will always be different, varying with waveforms and mean value (often times nonzero).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [Tutorial 1 as of commit `469219a`](https://github.com/SNL-WaterPower/WecOptTool/blob/8b03a19ccabcc47947f1fa06f8395972d8f6d2a2/examples/tutorial_1_wavebot.ipynb) with a WecOptTool version between v2.2.0 and v2.2.3.
2. In Part 2, edit the cell solving for the unconstrained average electrical power by modifying the following input line by adding the three lines of code that have the `###` commented:
```
results_2_nocon = wec_2_nocon.solve(
    waves, 
    obj_fun_2, 
    nstate_opt,
    scale_x_wec=scale_x_wec,  ###
    scale_x_opt=scale_x_opt,  ###
    scale_obj=scale_obj,      ###
)
```
3. Run the script in a Jupyter notebook.
4. Scroll down to the following inputs comparing the WaveBot in the constrained and unconstrained cases. The position and force plots for the unconstrained case (i.e. `wec_tdom_2_nocon['pos'].plot(label='unconstrained')` and `pto_tdom_2_nocon['force'].plot(label='unconstrained')`) will likely show a nonzero mean.
5. Rerun these two cells several times, and the plot will be different each time, even though the "Optimal average electrical power" displayed for the unconstrained case will always be -28.894±0.001 W.

**Expected behavior**
Multiple solves of the same unconstrained problem with correct scaling should yield the same optimal average electrical power *and* consistent, logical WEC kinematics. For example, the WaveBot example should have nonzero position and amplitude peaks somewhat higher than the constrained case:
[image]

Problems with incorrect scaling should struggle to converge; either yielding a suboptimal solution, taking a long time to converge to the optimal solution, or be unable to converge at all.

**Observed behavior**
The optimal average electrical power always converges to -28.894±0.001 W, but the WEC position and force displayed will have a different mean value each time. For example, the following two position plots were generated on consecutive runs:
[image]
[image]

**System:**
 - OS: Windows 10
 - Python version: 3.10.8
 - WecOptTool version: v2.2.3 (also observed with v2.1.0 and v2.0.0 by setting `nstate_opt = 2*nfreq+1` in Tutorial 1)
 - Any other relevant package version: N/A

**Additional information**
Running Tutorial 1 as-is (with the scaling parameters unset for the unconstrained case) will result in the expected behavior. This has been replicated for other models as well, but it is not confirmed whether this is universal.

cmichelenstrofer, 2023-03-31:
The issue is with the PTO force and its zero-frequency component being used in some parts of the code but not others. We are working on it.

dtgaebe, 2023-04-12:
@cmichelenstrofer, @gbacelli and me discussed this. The impedance matrix is not defined at zero frequency, so currently we're not considering the PTO force's zero frequency component $F_{p,0}$ in the standard objective function $P_e$. This should be considered since maintaining a constant force, means constant generator torque, which requires nonzero current and nonzero electrical power.

I think we could introduce a corrective term
```math
P_{e,0} = Re(Z_{22})(Re(Z_{12})^{-1}F_{p,0})^2
```
that captures the purely resistive caracteristics of the energy transduction for the zero component and subtract it from the transduced power. The assumption to derive this expression is that the velocity $U$ is zero.
```math
\begin{bmatrix}
F_{p} \\
Eff
\end{bmatrix}
=
\begin{bmatrix}
Z_{11} & Z_{12} \\
Z_{21} & Z_{22}
\end{bmatrix}
\begin{bmatrix}
U = 0 \\
Flo
\end{bmatrix}
```
For the case with an energy conversion from mechanical power to electrical power this equation are RI^2 losses. In 1D this looks like
```math
\begin{align}
\begin{bmatrix}
F_{p,0} \\
V_0
\end{bmatrix}
&=
\begin{bmatrix}
* & Re(Z_{12}) \\
* & Re(Z_{22})
\end{bmatrix}
\begin{bmatrix}
0\\
I_0
\end{bmatrix}\\
&=
\begin{bmatrix}
* & -NK_t \\
* & R_w
\end{bmatrix}
\begin{bmatrix}
0\\
I_0
\end{bmatrix}
\end{align}
```

```math
\begin{align}
P_{e,0} &= V_0  I_0 \\
&= (R_wI_0) I_0 \\
&= R_w(\frac{-1}{NK_t}F_{p,0})^2 \\
&= \frac{R_w}{(NK_t)^2}F_{p,0}^2
\end{align}
```

There might be a more elegant solution, like adding a zero frequency component to the impedance matrix, which would be purely Real. Then we might not need the corrective term.

dtgaebe, 2023-04-19:
[#221](https://github.com/SNL-WaterPower/WecOptTool/pull/221) created a bug when using a structured controller since we were using the first entry of `x_opt`, which is not necessarily the DC force. Instead of this corrective term we should add the zero frequency component to the `abcd` matrix. 
```math
\begin{align}
\begin{bmatrix}
I_{0} \\
V_0
\end{bmatrix}
&=
\begin{bmatrix}
Re(-Z_{12}^{-1}Z_{11}) & Re(Z_{12}^{-1}) \\
Re(Z_{21}-Z_{22}Z_{12}^{-1}Z_{11}) & Re(Z_{22}Z_{12}^{-1})
\end{bmatrix}
\begin{bmatrix}
0\\
F_{p,0}
\end{bmatrix}\\
&=
\begin{bmatrix}
* & \frac{1}{NK_t} \\
* & \frac{R_w}{NK_t}
\end{bmatrix}
\begin{bmatrix}
0\\
F_{p,0}
\end{bmatrix}
\end{align}
```

This will automatically add the desired effect

PR #221 (2023-04-14): Add power losses due to zero-frequency PTO force
## Description
Add the power losses due to zero-frequency PTO force. Derivation by Daniel. Fixes #213.

## Type of PR
- [x] Bug fix
- [ ] New feature
- [ ] Documentation
- [ ] Other: (specify)

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/SNL-WaterPower/WecOptTool/blob/main/.github/CONTRIBUTING.md)
- [x] The pull request is from an issue branch (not main) on *your* fork, to the [main branch in WecOptTool](https://github.com/SNL-WaterPower/WecOptTool).
- [x] The authors have given the admins edit access
 [... cut]
PR #227 (2023-04-24): Zero freq impedance power adjustment
## Description
Fixes #213 

## Type of PR
- [x] Bug fix
- [ ] New feature
- [ ] Documentation
- [ ] Other: (specify)

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/SNL-WaterPower/WecOptTool/blob/main/.github/CONTRIBUTING.md)
- [x] The pull request is from an issue branch (not main) on *your* fork, to the [main branch in WecOptTool](https://github.com/SNL-WaterPower/WecOptTool).
- [x] The authors have given the admins edit access
- [x] All changes adhere to the style guide including PEP8, Docstrings, and Type Hints.
 [... cut]
closed 2023-04-24 by dtgaebe