sandialabs/WecOptTool #223: Allow user to specify fmin

ryancoe, 2023-04-13:
We currently create the frequency vector using the fundamental (`f1`) and the number of frequencies (`nfreq`). However, this means that you can often wind up with a lot of low frequencies where no energy is present (see, e.g., `omega < 1` below). One could reduce the pseudo-spectral problem size by cutting out these lower frequencies.

[image]

https://github.com/SNL-WaterPower/WecOptTool/blob/9d8f4ef76c86cedf8ed997d85e1698f2bebc15b7/wecopttool/core.py#L1264-L1291

dtgaebe, 2023-08-23:
This will also be important to create the `WEC` `from_impedance`. I was looking into the frequency response functions data in fbWecCtrl of our pet WECs and `f_min` is 0.1Hz for both the WaveBot and FOSWEC. The fundamental of 0.1Hz is to coarse to capture the dynamics (especially the FOSWEC).

While the extrapolated impedance looked alright in a plot, I suddenly started having convergence issues, so we might want prioritize this enhancement even before V3 @cmichelenstrofer ?

cmichelenstrofer, 2023-11-01:
We will want to retain the mean (DC) component even if we set `fmin>1` . This will require some more effort...