sandialabs/WecOptTool #225: BUG: Last frequency component is wrong

cmichelenstrofer, 2023-04-15:
**Describe the bug**
The last frequency component of the results is not continues but has a significant jump.

**To Reproduce**
Run a case with irregular waves where the spectrum has significant ammount of energy in the last frequency (e.g. cut off the spectrum early). 

**Expected behavior**
Smooth tail.

**Observed behavior**
Large jump at last frequency.

**System:**
All.

**Additional information**
We suspect this is related to the DFT matrix.

cmichelenstrofer, 2023-04-15:
@michaelcdevin can you add more details?

michaelcdevin, 2023-04-17:
See attached zipped file for a basic example to replicate this behavior using WecOptTool v2.4.0 using a similar setup to Tutorial 1, part 1: [issue225_test.zip](https://github.com/SNL-WaterPower/WecOptTool/files/11253424/issue225_test.zip)

The highest frequency will give a different optimal excitation force each time as well. Since there is a large amount of energy in that frequency, this will substantially alter the optimal power:

Run 1:
[image]
**Optimal power:** -391.20254096338795 W

Run 2:
[image]
**Optimal power:** -398.47711760183483 W

michaelcdevin, 2023-04-17:
At least part of the issue here is that the `wot.td_to_fd` should not be multiplying the DC and Nyquist frequency by 2, since the two-sided wave spectrum would not have Hermitian symmetry at these frequencies for purely real signals (since we're removing the sine component of the Nyquist frequency). This is clear when looking at the effort in the frequency domain when calculating power by the top shared frequency being half the magnitude:
[image]

Fixing this corrects the effort, but the top frequency in the solution is still wrong, so I'm still troubleshooting this.

michaelcdevin, 2023-04-26:
**The root causes for this behavior have been identified:**
1) The truncations of the last row/columns from #191 now means the last row of the derivative matrix is all zeros, making the evaluated velocity (i.e. from `np.dot(time_mat, pos_td)`) at the last time step be zero no matter what, which should not be the case. This makes the last element of the WEC dynamics state vector have no effect on the solution. This causes:
* The last element of the dynamics state vector to be random, which is why the results at the top frequency are not smooth.
* The WEC inertia to be different (since the last element of the dynamics state vector is random), which is why the optimal power converges to a different solution. Note that the control state vector is never multiplied by the derivative matrix for unstructured controllers, so the PTO force will still result in a consistent value, which is why the solver still converges at all.

2) Not an explicit cause, but certainly bears mentioning: versions of WecOptTool prior to v2.2.0 (i.e. before #191) had a bug with the time vector creation that was masking this issue. This was due to there being one too many components in the time series making the time intervals incorrect, thus the top frequency was *not* actually the two-point wave. See the example code below using WecOptTool [v2.1.0](https://github.com/SNL-WaterPower/WecOptTool/releases/tag/v2.1.0). If the time vector is corrected in v2.1.0, the solution blows up since the last component of the control state vector has no effect on the PTO force (since the sine term at the last time step will always evaluate to zero) and nothing is constraining the velocity, so the last element of the WEC dynamic state vector exponentially increases at the last time step.

**Example of bug from cause (2) -- use WecOptTool v2.1.0:**
```
import numpy as np
import wecopttool as wot

nfreq = 10
f1 = 0.01
t = wot.time(f1, nfreq) # should be multiples of 5
w = wot.frequency(f1, nfreq, False)
wt = np.outer(t, w)
top_sin = np.sin(wt)[:, -1] # sine component of two-point wave, should be all zeros
print(f'Time vector: {t}')
print(f'Sine component of 2pt wave: {top_sin}')

# Time vector: [ 0.          4.76190476  9.52380952 14.28571429 19.04761905 23.80952381
# 28.57142857 33.33333333 38.0952381  42.85714286 47.61904762 52.38095238
# 57.14285714 61.9047619  66.66666667 71.42857143 76.19047619 80.95238095
# 85.71428571 90.47619048 95.23809524]
# Sine component of 2pt wave: [ 0.          0.14904227 -0.29475517  0.43388374 -0.56332006  0.68017274
# -0.78183148  0.8660254  -0.93087375  0.97492791 -0.9972038   0.9972038
# -0.97492791  0.93087375 -0.8660254   0.78183148 -0.68017274  0.56332006
# -0.43388374  0.29475517 -0.14904227]
```
**Point regarding the above comment:**
Given this new context, the correction of the DC and top frequencies not being multiplied by 2 is still correct in theory and should be fixed, but is not a reason for the top frequency being wrong.

michaelcdevin, 2023-05-05:
After further discussion, we determined the velocity being zero at the top frequency is actually correct behavior. Since $\omega t = n \pi$ for the Nyquist frequency, $v(f_{Nyquist}) = x'(f_{Nyquist}) \rightarrow \frac{\mathrm{d} }{\mathrm{d} t}(cos(\omega t)) = -\omega sin(\omega t) = 0$. Therefore, for unstructured controllers, the last frequency being arbitrary is the expected response, since the objective function of average power only considers velocity, not position or acceleration.

However, something is still not right here. I am still getting the same behavior when using a PI or PID controller, where position/acceleration should affect the objective function since it impacts the force.

Related bugs mentioned in this issue regarding the incorrect DFT matrix behavior and the 2nd derivative matrix affecting the acceleration have been addressed in #232.

cmichelenstrofer, 2023-05-05:
One clarification. The last component of velocity (last component in the state vectors is $cos(N \omega t)$ ) depends on the $sin(N\omega t)$ amplitude of the position which we do not include in our state (arbitrary since the $sin$ is always zero). So we need to make an assumption... and that assumption is that the amplitude of the sine component of position is zero (pure cosine).

michaelcdevin, 2023-05-05:
A bit more info on the behavior with a structured controller.

I'm testing convergence with a consistent `x_wec_0` and `x_opt_0` with a PI controller and 16 frequencies. Rerunning the same problem several times, the optimal `x_opt` has its first two elements different each time, and `x_wec` is completely different each time. Except for the first two elements, `x_opt` will converge to the same values even if `x_opt_0` and/or `x_wec_0` are changed.

The post processed results are the same as above, with an arbitrary last frequency and a different optimal average power each time.

michaelcdevin, 2023-05-06:
Disregard my earlier comment, I forgot to switch my script back to using the same waves each time...

The behavior I'm seeing with the PI controller w/ 16 frequencies is that `x_wec` and `x_opt` both converge to consistent values *except* for the last element of `x_opt`, which is unchanged from the last element in `x_opt_0`. Again, the PI controller means that the position of the Nyquist frequency should have an impact, but it isn't.

cmichelenstrofer, 2024-04-02:
# Power last frequency

## Objective function: Average mechanical power

The states are $X_{wec}$ and $F_{pto}$ which are the WEC position and
PTO force time series in the frequency domain, using an unstructured
controller for the force. The WEC velocity can be obtained as the
derivative of the position.

$$X_{wec} = \begin{bmatrix} x_{dc} \\ x_{a1} \\ x_{b1} \\ x_{a2}\end{bmatrix} \qquad $$

$$\dot{X}_{wec} = \begin{bmatrix} 0 \\ x_{b1}\omega_1 \\ -x_{a1}\omega_1 \\ 0 \end{bmatrix} \qquad$$
    
    \ddot{X}_{wec} = \begin{bmatrix} 0 \\ -x_{a1}\omega_1^2 \\ -x_{b1}\omega_1^2 \\
        -x_{a2} \omega_2^2
        \end{bmatrix} \qquad
        F_{pto} = \begin{bmatrix} f_{dc} \\ f_{a1} \\ f_{b1} \\ f_{a2}\end{bmatrix}

Note that the second derivative state can not be obtained by applying
the same derivative matrix to the velocity as was applied to the
position to obtain velocity. That is, the second derivative operator
(matrix) is not the first derivative operator squared; these differ in
the treatment of the last frequency component.

In the time domain, these are: $$\begin{split}
        x_{wec}(t) &= x_{dc} + x_{a1}\cos(\omega_1 t) + x_{b1}\sin(\omega_1 t) + x_{a2}\cos(2\omega_1 t) \\
        \dot{x}_{wec}(t) &= x_{b1}\omega_1\cos(\omega_1 t) + -x_{a1}\omega_1\sin(\omega_1 t) \\
        f_{pto}(t) &= f_{dc} + f_{a1}\cos(\omega_1 t) + f_{b1}\sin(\omega_1 t) + f_{a2}\cos(2\omega_1 t) 
        \end{split}$$

The power in the time domain is the product of velocity and force. This
is equivalent to a convolution integral in the frequency domain.
$$\label{eq:mechanicalpower}
        \begin{split}
        p(t) = &\dot{x}_{wec}(t)\cdot f_{pto}(t) \\
        = &\quad f_{dc}x_{b1}\omega_1\cos(\omega_1 t) \\ 
        &+ f_{a1}\cos(\omega_1 t)x_{b1}\omega_1\cos(\omega_1 t)\\ 
        &+ f_{b1}\sin(\omega_1 t)x_{b1}\omega_1\cos(\omega_1 t) \\
        &+ f_{a2}\cos(2\omega_1 t)x_{b1}\omega_1\cos(\omega_1 t) \\
        &- f_{dc}x_{a1}\omega_1\sin(\omega_1 t) \\ 
        &- f_{a1}\cos(\omega_1 t)x_{a1}\omega_1\sin(\omega_1 t)\\ 
        &- f_{b1}\sin(\omega_1 t)x_{a1}\omega_1\sin(\omega_1 t) \\
        &- f_{a2}\cos(2\omega_1 t)x_{a1}\omega_1\sin(\omega_1 t) \\
        \end{split}$$

The average power is the integral of power in the time domain, or the DC
component of power in the frequency domain. Using orthogonality of
sinusoids, many of these terms are zero. Similarly, the integral of a
sine or cosine over an integer number of periods is zero.

$$\begin{split}
        \bar{P}_{mech} = &\int_0^{2\pi/\omega_1} p(t) \mathop{}\!\mathrm{d}{t} \\
        = &\int_0^{2\pi/\omega_1}  \biggl[ \\
        % &\qquad (f_{dc}x_{b1}\omega_1)\cos(\omega_1 t)\\
        &\quad+ (f_{a1}x_{b1}\omega_1)\cos(\omega_1 t)\cos(\omega_1 t) \\
        % &\quad- (f_{dc}x_{a1}\omega_1)\sin(\omega_1 t) \\
        &\quad- (f_{b1}x_{a1}\omega_1)\sin(\omega_1 t)\sin(\omega_1 t) \\
        &\biggr]\mathop{}\!\mathrm{d}{t}\\
        = &\pi(f_{a1}x_{b1} - f_{b1}x_{a1})
        \end{split}$$

If we generalize to $n$ frequencies:

$$\label{eq:results}
        \bar{P}_{mech} = \pi\left( \sum_{k=1}^{n-1} k \left(f_{ak}x_{bk} - f_{bk}x_{ak} \right) \right)$$

**Neither the DC component nor the last frequency component of the PTO
force or the WEC position have any effect on mechanical power.**

# Dynamics
[... 188 lines cut ...]
    DF_p &= (D_a+{i\mkern 1mu}D_b)(f_a + {i\mkern 1mu}f_b) \\
             &= D_a f_a - D_bf_b + {i\mkern 1mu}(D_af_b + D_bf_c)
    \end{split}\end{aligned}$$ $$\begin{aligned}
    \begin{split}
        I = \begin{bmatrix}
        I_a \\
        I_b 
    \end{bmatrix}
    = 
    \begin{bmatrix}
    A_a\dot{X}_a - A_b\dot{X}_b + B_a f_a - B_bf_b \\
    A_a\dot{X}_b + A_b\dot{X}_c + B_af_b + B_bf_c
    \end{bmatrix}
    \end{split} \\
    \begin{split}
        V = \begin{bmatrix}
        V_a \\
        V_b 
    \end{bmatrix}
    = 
    \begin{bmatrix}
    C_a\dot{X}_a - C_b\dot{X}_b + D_a f_a - D_bf_b  \\
    C_a\dot{X}_b + C_b\dot{X}_c + D_a f_a - D_bf_b
    \end{bmatrix}
    \end{split}   \end{aligned}$$

cmichelenstrofer, 2024-04-02:
Through this issue we 
* fixed the DFT matrix
* use a different second derivative operator

The last issue we were discussing is not a bug and is actually the expected behavior. See the following writeup showing how for the linear problem with average mechanical power as the objective and an unstructured controller the DC and Nyquist components of the WEC position and PTO force are arbitrary.

closed 2024-04-02 by cmichelenstrofer