sandialabs/WecOptTool #229: A question about the Pseudo-spectral methods

smallduan, 2023-04-24:
Hi 
I want to learn more about the Pseudo-spectral methods which is used in WecOptTool, but I didn't find the details in the references. Please give me some information (paper or book) to understand how this method works for time domain calculations of wave energy devices.

Best regards
smallduan

smallduan, 2023-04-26:
Hi 
When performing multi-float calculations, the difference in motion between multiple floats is affected by the initial phase of the motion, and I cannot determine the accuracy of the displacement difference in the time domain for the results obtained from the calculations. Can you tell me the source formula for this part of the algorithm.

Best regards

dtgaebe, 2023-05-01:
Hi @smallduan,

This reference page: https://snl-waterpower.github.io/WecOptTool/references.html#id2 didn't have what you're looking for?

Could you open an additional issue for your comment and please share some more details? Please feel free to share code or files as well.

Best,
Daniel

smallduan, 2023-05-09:
@dtgaebe
Thank you for your answer
I am so sorry
I didn't read the documentation properly and I'm sorry for wasting your valuable time.

Best regards
smallduan

closed 2023-05-15 by ryancoe