sandialabs/WecOptTool #230: CI: Hyperlink requests to IEEExplore site returns HTTP status 418

michaelcdevin, 2023-04-26:
In the past week, requests to [https://ieeexplore.ieee.org](https://ieeexplore.ieee.org) have bizarrely started to return [HTTP status code 418](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/418). This causes an issue in WecOptTool since two of the theory references ([doi:10.1109/TSTE.2014.2371536](https://doi.org/10.1109/TSTE.2014.2371536) and [doi:10.1109/9.467672](https://doi.org/10.1109/9.467672)) redirect to the IEEExplore website, and so throws an exception in our CI (see [here](https://github.com/SNL-WaterPower/WecOptTool/actions/runs/4770176846/jobs/8481216087) and [here](https://github.com/SNL-WaterPower/WecOptTool/actions/runs/4788206873/jobs/8514495583)).

These two references open fine when opening in a browser (tested with Firefox, Chrome, and Edge), and work with cURL without issue.

To replicate from a Python prompt:
```
import requests
requests.get('https://ieeexplore.ieee.org/').status_code
# returns 418
```

michaelcdevin, 2023-04-26:
It looks like this is a case where the status code 418 is a stand-in for a `403 Forbidden` status, as explained [here](https://en.wikipedia.org/wiki/Hyper_Text_Coffee_Pot_Control_Protocol#Usage) (I don't have the slightest idea why this happens and can't find any reasoning online). Regardless, it is being caused by the IEEExplore site suddenly blocking some user agents (again, no idea why). Spoofing a general user agent seems to fix the problem:

```
import requests
requests.get('https://ieeexplore.ieee.org/', headers={'User-Agent': 'my-app/0.0.1'}).status_code
# returns 200
```

closed 2023-04-26 by michaelcdevin

hugopm123, 2026-11-03:
Con el agente de usuario, te dejo de dar ese problema?

PR #231 (2023-04-26): Add custom user agent when linking to IEEExplore
Resolves #230 (plus some minor cleanup of the `linkcheck_ignore` list)