sandialabs/WecOptTool #239: BUG: SSLError in Sphinx `linkcheck` behind company proxy

michaelcdevin, 2023-05-17:
**Describe the bug**
See https://github.com/sandialabs/WecOptTool/pull/169#issuecomment-1551645986. This only seems to occur behind a company proxy, and is not replicated in the GitHub Actions CI.

**To Reproduce**
1. Behind an internet proxy, run `python docs/build_docs.py` from the WecOptTool repo on the `main` branch

**Expected behavior**
No SSLError is encountered, even when behind a proxy.

**Observed behavior**
The following error occurs during the `linkcheck` step:
```
broken link: https://github.com/HIPS/autograd (HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /HIPS/autograd (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)'))))
```

**System:**
 - OS: Observed on Windows and Mac
 - Python version: v3.10.10
 - WecOptTool version: v2.4.0

ryancoe, 2023-10-27:
@michaelcdevin - I think we can probably close this now that you've completed #287. Do you agree?

michaelcdevin, 2023-10-27:
I believe the SSLError would still occur, but it would just be a warning now. This could just be checked in the CI though, so this is super low priority and probably not worth the effort. I'll go ahead and close it.

closed 2023-10-27 by michaelcdevin