sandialabs/WecOptTool #253: residual function

cmichelenstrofer, 2023-07-11:
The hydrodynamic bodies's residual constraint is built in the `core` solver. One can recreate it with the post-processed time-series, but it might be useful to just supply a function that does this calculation. The main residual (and any other equality constraint) could also be printed during the optimization output.

ryancoe, 2023-10-06:
@cmichelenstrofer - I'm a bit confused by your comment above. The _scaled_ residual function is indeed created in `wot.solve`, 

https://github.com/sandialabs/WecOptTool/blob/6c9b8524d80c1a4469d12b7d1a0b453089435458/wecopttool/core.py#L766-L770

but there is a hidden function for the residual equation that a user can call

https://github.com/sandialabs/WecOptTool/blob/6c9b8524d80c1a4469d12b7d1a0b453089435458/wecopttool/core.py#L604-L612

cmichelenstrofer, 2023-10-09:
@ryancoe yes, you are right! I suggest we make it a regular function (not a hidden one). And check that we can use it (e.g. as `callback`) to print out the value during the solve. This is something @gbacelli has asked for a few times. Should give us insight into the convergence of the optimization.

cmichelenstrofer, 2023-10-11:
* changed the residual function to be non-hidden.
* fixed a bug with `callback` (state x was not being scaled)

To print all the residuals (each DOF at each time step) you can do something like this: 

```python
def callback(wec, x_wec, x_opt, waves):
    wot._log.info(f"dynamic equation residual: {wec.residual(x_wec, x_opt, waves)}")
    return 
```

or to just check if it is satisfied:

```python
def callback(wec, x_wec, x_opt, waves):
    wot._log.info(f"dynamic equation residual: {np.allclose(wec.residual(x_wec, x_opt, waves), 0)}")
    return 
```

The callback is called *after* each iteration. I tried it with Tutorial 1 and the condition is satisfied after the first iteration (never prints `false`).  

Similarly the callback could include user defined constraints functions.

commit 21c1eccd (2023-10-12): Make residual function more accessible. Closes #253 (#276)
closed 2023-10-12 by cmichelenstrofer