sandialabs/WecOptTool #263: Feature request: Add integration + documentation language for Mamba

michaelcdevin, 2023-07-27:
**Feature description.**
[Mamba](https://mamba.readthedocs.io/en/latest/index.html) is an Python environment manager based on [Conda](https://docs.conda.io/en/latest/). While it is newer and less established than Conda, it is well-maintained and has a quickly growing user base. Mamba also has far superior package resolution compared to Conda -- testing it out on my personal computer, `conda install wecopttool` took over 30 minutes to resolve dependencies (unusually long, but in the past has still taken 5-10 minutes), while `mamba install wecopttool` took about 90 seconds.

We don't reference Mamba at all in the installation instructions for WecOptTool, but I think it would be wise to, as I only see Mamba continuing to gain popularity in the future. We should also check to make sure `mamba install wecopttool` works as expected -- Mamba generally works seamlessly with `conda-forge`, but it is known to still have bugs that are not present in Conda.

**Issue addressed.**
Support of WecOptTool installation via Mamba.

**Describe the solution you'd like**
Allow users to run `mamba install wecopttool` without issue, much as they would for `conda install wecopttool`. Also include related documentation: an "Option 3" in the Getting Started section of the README and updated "Installation for developers" in CONTRIBUTING.md. If Mamba works *really* well, we could even recommend Mamba over the other installation options.

**Describe alternatives you've considered**
N/A

**Interest in leading this feature development?**
I can lead this

saltynexus, 2023-08-29:
I agree...it takes forever to install via "classic" conda. My install was "hanging" in "solving environment" for a long time.

Take a look at installing [libmamba](https://conda.github.io/conda-libmamba-solver/getting-started/) as the solver. Once installed, move to your virtual environment and do

`conda install -c conda-forge wecopttool --solver=libmamba`

This took about 2 min to install.

michaelcdevin, 2023-08-29:
This is great @saltynexus! I was unaware of libmamba before now. I will continue looking at Mamba too, but using Conda with the libmamba solver is a major improvement. This may be worth mentioning in WecOptTool install instructions.

ryancoe, 2023-09-22:
On my M1 mac and I couldn't get `libmamba`, `mamba`, or `conda` to work (i.e., can't do `conda install -c conda-forge wecopttool --solver=libmamba`, but can I can do `pip install .` I did start using `mamba` over `conda` and it seems nice. @michaelcdevin - Could you see if either of the following work for you in Windows?

- `conda install -c conda-forge wecopttool --solver=libmamba`
- `mamba install -c conda-forge wecopttool`

saltynexus, 2023-09-22:
FYI...during my search to resolve this, I read a new version of conda is supposed to released by the end of this month (September) with the default solver being that used in mamba. So this should resolve the issue moving forward across the board for conda installs. Also, just to add to @ryancoe's message, I'm using RHL as an OS. I did have some issues following an update to the latest release of libmamba (1.5) causing "Entry point" errors, but was able to resolve by reinstalling libmamba.

michaelcdevin, 2023-09-22:
@ryancoe I was able to get `conda` working using the `libmamba` installer by first installing it using the method listed on the [documentation page](https://conda.github.io/conda-libmamba-solver/getting-started/), then installing WecOptTool normally:

```
conda install -n base conda-libmamba-solver
conda config --set solver libmamba
conda install -c conda-forge wecopttool
```

I have been able to get `mamba install -c conda-forge wecopttool` to work on my home computer, but not my work laptop (both are Windows) due to an SSLError. There seems to be some proxy issue with `mamba` that I haven't been able to figure out the correct configuration for that isn't present (or just has been configured correctly) with `conda`.

ryancoe, 2023-10-05:
To avoid getting too deep into this, I've decided in #267 to just mention that you can use `mamba` instead of `conda`. I think offering up the `libmamba` option in our `README` will likely confuse more people than it helps.

michaelcdevin, 2023-10-05:
I agree with that approach, especially since it sounds like `conda` will default to using the `libmamba` solver in the near future regardless, as per @saltynexus.

michaelcdevin, 2023-10-06:
For what it's worth, `mamba install wecopttool` now works on my work laptop (i.e., through a company proxy) after updating `mamba` to its most recent version with `mamba update -n base mamba`, so it looks like a recent version (v1.5?) resolved the `conda config` not carrying over.

PR #267 (2023-10-13): Documentation improvements/updates
Minor improvements/updates to the documentation inspired by some recently published papers.

- [x] Closes #263 
- [x] Closes #268
- [x] Add references to recently published papers
closed 2023-10-13 by michaelcdevin