sandialabs/WecOptTool #268: Link to "companion notebooks" and papers in relevant tutorials

ryancoe, 2023-09-08:
A number of our [Tutorials](https://sandialabs.github.io/WecOptTool/tutorials.html#) correspond to journal or conference papers, but sometimes the papers have a bit more detail than the the tutorials. Additionally, the papers are produced with a certain version of WecOptTool, but we don't necessarily want to commit to updating analyses in the papers as the WecOptTool source evolves. @cmichelenstrofer and @dtgaebe have addressed this by creating dedicated repositories for _companion notebooks_ to the papers, which are sometimes (always?) referenced from the papers themselves.

- https://github.com/cmichelenstrofer/IEEE_TSTE_2023
- https://github.com/cmichelenstrofer/EWTEC_2023
- https://github.com/dtgaebe/OMAE_2023_103899

I'd suggest that we update the markdown in the relevant tutorials to reference both the appropriate paper and companion notebook. For example:

 _"This device was studied in [[5]](https://doi.org/10.1109/TSTE.2023.3272868); a companion notebook that is similar to this tutorial, but directly replicates the analysis in [[5]](https://doi.org/10.1109/TSTE.2023.3272868) is available at https://github.com/cmichelenstrofer/IEEE_TSTE_2023."_

cmichelenstrofer, 2023-10-02:
we should also have a single list somewhere with these published studies using WecOptTool and links to the papers and notebooks.

ryancoe, 2023-10-03:
I agree. I started to do what I suggested above (put links to appropriate papers/repos into our tutorials), but I stepped away from it because it let like something that'd be pretty hard to maintain. I prefer a central location, e.g., the [References page](https://sandialabs.github.io/WecOptTool/references.html).

PR #267 (2023-10-13): Documentation improvements/updates
Minor improvements/updates to the documentation inspired by some recently published papers.

- [x] Closes #263 
- [x] Closes #268
- [x] Add references to recently published papers
closed 2023-10-13 by michaelcdevin