sandialabs/WecOptTool #269: Feature request: example of saturated pi controller

cmichelenstrofer, 2023-09-28:
**Feature description.**
A clear and concise description of the proposed feature. 

**Issue addressed.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...] Link to any open [issues](https://github.com/sandialabs/WecOptTool/issues) this would address.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.  Consider: why does this belong in WecOptTool? Is it general or specific to my problem? Would it be beneficial to others? 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered. Also explain any current workarounds.

**Interest in leading this feature development?**
Whether you are interested in leading or helping with the development of this feature. 

**Additional information**
Add any other context or screenshots about the feature request here.

cmichelenstrofer, 2023-10-02:
Here's a noteboook that uses a saturated PI controller, using `np.clip`. 

[saturated_pi.ipynb.zip](https://github.com/sandialabs/WecOptTool/files/12786573/saturated_pi.ipynb.zip)

The scales are not ideal yet... takes many iterations to converge and sometimes it needs to be run twice. But it shows how to create a saturated PI controller and that it works. 

Interestingly, the optimal gains are not the same as for the PI (since the position and velocities are not the same!). 

Using a max force of +/-1000N, this is a regular wave result:
[image]

and irregular:
[image]

The saturated PI controller is achieved as 

```python
f_max = 1000

def saturated_pi(pto, wec, x_wec, x_opt, waves=None, nsubsteps=1):
    force_td_tmp = wot.pto.controller_pi(pto, wec, x_wec, x_opt, waves, nsubsteps)
    force_td = np.clip(force_td_tmp, -1*f_max, f_max)
    return force_td
```

cmichelenstrofer, 2023-10-02:
We should probably include this as either its own controller or a keyword on the existing PID controllers. @ryancoe @dtgaebe FYI

ryancoe, 2023-10-03:
Nice work! If we can do it cleanly (and I think we can), I'd prefer to integrate it into the existing function(s).

At a higher level, it'd be really great to have a tutorial where we illustrate a good workflow for device and control design (i.e., start with unstructured, work towards something you can implement in real-time). The Pioneer WEC might offer a good oppurtunity for this.

cmichelenstrofer, 2023-10-06:
Regular  wave results (more plots):

[image]

[image]

cmichelenstrofer, 2023-10-06:
irregular: 

[image]

[image]

cmichelenstrofer, 2023-10-06:
@ryancoe @dtgaebe I added some more results above.

cmichelenstrofer, 2023-10-06:
I'm working now on making the PI controller general for multiple PTO DOF.

ryancoe, 2023-10-06:
Ok, you may want to talk with @dtgaebe about multi-dof feedback controllers if you're able

ryancoe, 2023-10-06:
@cmichelenstrofer - Can you add a third curve to [your regular wave comparison above](https://github.com/sandialabs/WecOptTool/issues/269#issuecomment-1751282967): unstructured controller with a force constraint applied

cmichelenstrofer, 2023-10-06:
@ryancoe The results are (nearly) identical to the saturated PI!! Maybe we shouldn't include a saturated PI force then... 

[image]

[image]

ryancoe, 2023-10-06:
Very cool! They'll only be the same for a regular (monochromatic) excitation like you have here, so I think that it is useful to add this functionality. However, this further emphasizes to me the need to have some good guidance on when/how to use these different controllers in the design process.

cmichelenstrofer, 2023-10-06:
Ah right! 

It is hard to tell which is faster. The unstructured + constraints converged in only 11 iterations, while the saturated PI took 76... but that might just be because of the scaling. Total time divided by number of iterations, the saturated pi controller is about twice as fast.

rebeccamccabe, 2023-10-07:
For a better initial guess, instead of using the unsaturated amplitude Fp, the fundamental of the saturated waveform will have an amplitude of alpha * Fmax / Fp, where alpha is a value ranging from 1 (no saturation) to 4/pi (full saturation, the fundamental of a square wave): 
[image]
(from the Franklin-Powell-Emami Control textbook section on describing functions)

To go even further (unclear this is actually necessary) I've also derived the math for the higher harmonics too, which I think should be exact for a linear problem.
[image]
which is derived here https://github.com/symbiotic-engineering/MDOcean/blob/rgm-force-saturation/dev/describing_fcn_error.mlx

rebeccamccabe, 2023-10-07:
other possible idea to help with convergence of the saturated controller - use a [sigma approximation](https://en.wikipedia.org/wiki/Sigma_approximation) which alleviates the overshoots at the corners, and should mean you need fewer harmonics to stay within the constraint.

cmichelenstrofer, 2023-10-09:
> which is derived here https://github.com/symbiotic-engineering/MDOcean/blob/rgm-force-saturation/dev/describing_fcn_error.mlx

Really cool @rebeccamccabe! @ryancoe if we publish an example using a saturated PI, we should use this as initialization. @dtgaebe maybe this will help with the case you were running?

cmichelenstrofer, 2023-10-09:
> use a [sigma approximation](https://en.wikipedia.org/wiki/Sigma_approximation)

This would be neat to try! Probably only needed in the PTO force, since that's the only one with discontinuities.

cmichelenstrofer, 2023-10-09:
> Ok, you may want to talk with @dtgaebe about multi-dof feedback controllers if you're able

@ryancoe, I talked to @dtgaebe; I will address off-diagonal multi-DOF feedback controllers in this same PR. We should also consider an example with the FOSWEC (with 2-DOF experimentally-derived impedance).

cmichelenstrofer, 2023-10-09:
Regarding the *sigma approximation*: The way I'm doing the saturated PID I take the responses (positions, velocities, accelerations) which are not discontinuous, multiply by the respective gains, and sum. I then clip the force using `np.clip`. The harmonics of the force are not directly represented. 

It could be used, however, for the unstructured controller with constraints. Or for the state (position) when there are state constraints. I suggest we provide a different `tmat` that does the sigma approximation in addition to the one we have that does the pure Fourier sum. If we want to do this we should start a new issue.

cmichelenstrofer, 2023-10-11:
Test cases: 
[saturated_PI_dev.zip](https://github.com/sandialabs/WecOptTool/files/12873678/saturated_PI_dev.zip)

I have been getting strange results. Let's discuss in next meeting.

cmichelenstrofer, 2023-10-12:
I did a test case with the WaveBot in 2 DOF (heave-surge), with two PTOs, and regular wave. I ran with

- Full 2x2 PI gains (4*2 variables) (*2 because there are two gain matrices, one for velocities and one for positions)
- Symmetric PI gains (3*2 variables)
- Diagonal PI gains (2*2 variables)

They all give the same results (same as optimal, unstructured) but the gain matrices are different. Which means the optimal gain matrix is not unique. 

@ryancoe does this seem right? 

> Heave gains (assymetric):
> 
> [[-6895.13315227,   476.57829602]
>  [    9.43973362,    -2485.50637345]]
> 
> Heave gains (symetric):
> 
> [[ 1845.3334533 ,  -5443.49331145]
>  [-5443.49331145,  2526.3638102 ]]
> 
> Heave gains (diagonal):
> 
> [[-4407.2907246,      0.                        ]
>  [    0. ,                        -1083.72205073]]

and 

> Surge gains (assymetric):
> 
> [[ 2921.33289821,  5694.14061427]
>  [-7092.73108451,  891.82007449]]
> 
> Surge gains (symetric):
> 
> [[ 2.29386447e+03, -2.58101173e-08]
>  [-2.58101173e-08, -4.42551910e-16]]
> 
> Surge gains (diagonal):
> 
> [[ 8884.82149942,     0.        ]
>  [    0. ,        -3805.42951812]]

[saturated_PI_dev.zip](https://github.com/sandialabs/WecOptTool/files/12885931/saturated_PI_dev.zip)

PR #273 (2023-11-10): Saturated feedback controllers
## Description
This PR
*  adds the ability to saturate (symmetric or asymmetric max and min forces) the PTO force from PID controller. Closes #269 

## Type of PR
- [x] New feature

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/sandialabs/WecOptTool/blob/main/.github/CONTRIBUTING.md)
- [x] The pull request is from an issue branch (not main) on *your* fork, to the [main branch in WecOptTool](https://github.com/sandialabs/WecOptTool).
- [x] The authors have given the admins edit access
- [x] All changes adhere to the style  [... cut]
closed 2023-11-10 by cmichelenstrofer