sandialabs/WecOptTool #270: Test recent Python version compatibility

michaelcdevin, 2023-10-05:
[Python 3.12 was released three days ago](https://www.python.org/downloads/release/python-3120/). Currently [we still specify users to use Python 3.9 and 3.10 only](https://github.com/sandialabs/WecOptTool/blob/main/README.md#getting-started), though Python 3.11 seems to work seamlessly in my experience on Windows. We should confirm that 3.11 works across platforms and see whether 3.12 works as well, and update the README if so

ryancoe, 2023-10-06:
Would we do that by updating this line?

https://github.com/sandialabs/WecOptTool/blob/e6cda25fd4fed39e2354237b75417c7f9fee03be/.github/workflows/push.yml#L22

michaelcdevin, 2023-10-06:
@ryancoe that should work for Python 3.11. It looks like a number of dependencies still specify `python <3.12`, so I don't think this would work with 3.12 quite yet.

michaelcdevin, 2023-10-11:
Python 3.11 seems to work with the CI, will be updated in #278. It will probably be awhile before all of our dependencies are supported by 3.12, but I will monitor this and update when they are.

michaelcdevin, 2025-01-14:
Looks like all of our dependencies now work even with Python 3.1**3** and all the tests are passing... sorry for letting this fall through the cracks for so long! I created #390 to close this.

jtgrasb, 2025-01-30:
Python 3.13 is still not compatible with wavespectra so did not work. But, we did update to 3.12 in #390 and #394. This issue is resolved.

closed 2025-01-30 by jtgrasb