sandialabs/WecOptTool #272: Feature request: Convex Solver

rebeccamccabe, 2023-10-06:
**Feature description.**
In cases where the optimization is convex, use a convex programming interface such as CVXPy or GPKit instead of the current scipy.optimize.minimize SLSQP solver. This can speed up the solve, provide parameter sensitivities automatically, and guarantee a globally optimum solution. 

**Issue addressed.**
For the standard basic optimization in wecopttool, the objective is quadratic (power), the inequality constraints are linear (force and position maximums), and the equality constraint is linear (linear dynamics), so it is a quadratic program. But even in some cases of nonlinear dynamics and constraints, the problem can still be convex via geometric programming. ie if nonlinear quadratic drag or other second order terms are an added force, as long as the coefficients are positive then the dynamics equality constraint becomes a posynomial instead of linear, which can be solved as a convex geometric program with the "posynomial equality relaxation" method (see https://people.eecs.berkeley.edu/~pabbeel/papers/2012_gp_design.pdf). 

**Describe the solution you'd like**
Integrate a convex solver into core.py. Convex solvers can identify when the provided problem isn't convex, so it perhaps it makes sense to try the convex solver by default, and if it isn't convex then try a nonlinear solver like SLSQP. Ideally also the sensitivity feature of these solvers would be utilized (ie https://www.cvxpy.org/tutorial/advanced/index.html#sensitivity-analysis-and-gradients, https://gpkit.readthedocs.io/en/latest/gettingstarted.html#sensitivities-and-dual-variables). 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered. Also explain any current workarounds.

**Interest in leading this feature development?**
I can potentially work on this but probably not until the spring or summer.

cmichelenstrofer, 2023-10-09:
related #12

rebeccamccabe, 2023-11-26:
I tried out a toy problem for power optimization in CVXPy and realized that I had been misunderstanding the conditions of convex programming. While the expression for power is log-log-convex as I expected, convex programming only applies to _minimization_ of convex functions, or maximization of concave functions. Power optimization is the opposite (max convex, or min concave if you negate the objective), so CVXPy and other convex solvers can't handle it. 
Here's my notebook trying it out: https://github.com/symbiotic-engineering/WEC-DECIDER/blob/rgm-optim-formulation/convex_attempt.ipynb
It seems like there are other constraints that can be applied to enforce convexity (described more in the notebook), but I'd need to dig deeper before being sure that's worthwhile. Even if it's nonconvex, it still might be useful to leverage the quadratic program structure, which should be faster and perhaps more robust than the ipopt/slsqp NLP solvers but wouldn't have the benefits of convexity like free sensitivities or global optimality without an initial guess.

rebeccamccabe, 2023-11-27:
Sorry I think my previous comment was incorrect. The toy problem I constructed was too simplified (not incorporating any wave dynamics) that it changed the convexity properties substantially and was misleading. Looking back at Giorgio's Numerical Optimal Control paper, the actual objective is a convex quadratic so the problem is guaranteed to be convex for as long as the constraints (dynamics) are linear.

rebeccamccabe, 2023-11-30:
[WEC_DECIDER.pdf](https://github.com/sandialabs/WecOptTool/files/13510436/WEC_DECIDER.pdf)
Here is a writeup with some thoughts on convexity slightly more formalized. I am very open to feedback since I am new to convex programming.