sandialabs/WecOptTool #293: Tutorial 1 inconsistent force constraints

cmichelenstrofer, 2023-11-01:
@dtgaebe pointed out that in tutorial 1 we have the following discussion: 

> The attentive user might have noticed that the amplitude of the position, force and power signals is about half the magnitude of the signals we plotted in the first part of the tutorial. We can see that optimizing for electrical power requires optimal state trajectories with smaller amplitudes. For most WECs the electrical power is the usable form of power, thus the WEC should be designed for electrical power and we can avoid over-designing, which would results from expecting the forces associated with the optimal trajectories for mechanical power maximisation.

but the two cases have different force constraints, so we are not really comparing apples to apples.

dtgaebe, 2024-01-23:
I gave this some thought. I think we shouldn't be comparing apples to apples. Instead we wanted to illustrate the optimal solution for different forms of power subject to slight constraints, which we do.

I would go even further and illustrate how to constrain the generator current instead of the linear PTO force for the second part of the tutorial (electrical power).

```python
gen_current_max = 7.5
def const_f_pto(wec, x_wec, x_opt, waves):
    current, _ = pto.power_variables(wec, x_wec, x_opt, waves, nsubsteps)
    return gen_current_max - np.abs(current.flatten())

```

We can add an explanation that this approximately corresponds to a linear force constraint of `f_max = gear_ratio*torque_constant*gen_current_max` and also state that the actual generator max current is 20A (if I recall correctly)

On the other hand, one might raise the question if this is what we want to illustrate and highlight...

michaelcdevin, 2024-10-18:
Resolved in #339

PR #339 (2024-04-23): Tutorial cleanup
## Description
This is a major PR with various edits of the four tutorials included in the [`examples`](https://github.com/sandialabs/WecOptTool/tree/dev/examples) directory. This PR includes the following changes:

* Tutorial 1 is substantially reworked to make it easier to follow
  * Part 1 is reordered to define the wave conditions, then the WEC object, then the objective function in sequence. Currently, Tutorial 1 starts defining the WEC object, switches to defining the wave condition, then goes back to the WEC object, which can be difficult to follow.
  * More details are [... cut]