sandialabs/WecOptTool #296: cleanup linear damping functions

cmichelenstrofer, 2023-11-15:
We have three different (but related) functions: 

* `add_friction`  to add additional linear friction to the BEM data xarray. 
* `check_linear_damping` to modify the friction if radiation damping + friction is negative or close to zero. This modifes the BEM xarray
* `check_impedance` is the same but for an impedance array instead of a BEM xarray (i.e. one is used for `WEC.from_bem()` and the other for `WEC.from_impedance()`.  

These can be confusing. Suggested changes:

- [ ] `check_linear_damping` call `add_friction`
- [ ] consider renaming the functions to make it more obvious what they do
- [ ] have these functions always modify `friction` and `radiation_damping`, even in the case of frequency-dependent changes. Note that `frequency` can be frequency-dependent.

michaelcdevin, 2023-11-15:
>  have these functions always modify friction and radiation_damping, even in the case of frequency-dependent changes. Note that frequency can be frequency-dependent.

Point of clarity, the code currently makes it so `friction` is frequency independent, i.e. it only has dimensions of `[radiating_dof, influenced_dof]`. I don't think it is too difficult to change this in `add_linear_friction`, but we should be sure to update docstrings in functions where `friction` is an input when we do so -- `from_bem` and `from_floating_body` come to mind.

cmichelenstrofer, 2023-11-15:
Early on I tried to make `friction` work whether it was [ndof, ndof] or [ndof, ndof, nfreq]. I thought that was still the case, but I just checked `core` and `test_core` and this functionality is not there... at least not docuemented or tested. But I think we just add `frequency` to `radiation_damping` which is [ndof, ndof, nfreq]. The summation works even if `friction` doesn't have the third dimension, but should obviously work if it does. The places it might not work is where we create/add friction to a `bem_data` that doesn't have it. 

I think making this functionality official is the best option. Hopefully it isn't too difficult.

michaelcdevin, 2023-11-15:
> But I think we just add frequency to radiation_damping which is [ndof, ndof, nfreq]

I think you're right, friction is only ever added to radiation damping, so it should work regardless of whether it has the `omega` dimension or not. It should just be a matter of changing the data_var when we add `friction` in `add_linear_friction` to my understanding.

(also, the order is [nfreq, ndof, ndof] now ;) )

cmichelenstrofer, 2023-11-15:
> (also, the order is [nfreq, ndof, ndof] now ;) )

oops 😅

cmichelenstrofer, 2024-04-02:
There's no easy way to combine these functions, they have different uses. But I renamed the variables to make it more clear.

PR #328 (2024-04-02): damping naming and consistently change radiation damping
Closes #296
closed 2024-04-02 by cmichelenstrofer